OrcaSlicer/OrcaSlicer #5064: Some unknown/unnecessary perimeters appearing.

rednaxal, 2024-04-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

Open the attached project. Slice it. See it. 

### Actual results

For whatever reason the slicer adds some unwanted/unexplained/unnecessary internal perimeters on some flat surfaces:
[image]
[image]

### Expected results

It should not add those strange unnecessary internal perimeters at all. The vanilla Prusa Slicer 2.7.4 is not having this issue. 
The following nice and clean result is from Prusa Slicer 2.7.4
[image]

### Project file & Debug log uploads

[hex SC__trading_yellow.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/15016475/hex.SC__trading_yellow.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-04-18:
I confirm strange behavior. 
When I choose different printer profile it's slice ok.

Jeevus3D, 2024-04-19:
Same issue here. v2.0 - issue seems to be caused by "Only one wall on top surfaces" setting. With the setting disabled, the perimeter issue goes away regardless of arachne or classic generation.

igiannakas, 2024-04-22:
Set your single line top detection to 100%. It’s 300% by default and on some models it causes this artefact.

rednaxal, 2024-04-22:
As @igiannakas mentioned to get rid of this defect it is necessary to lower the value of the line top detection.

closed 2024-04-22 by rednaxal