OrcaSlicer/OrcaSlicer #5065: Wiki not public

zimivanende, 2024-04-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

/

### Printer

bambu

### How to reproduce

click on smooth spiral 

### Actual results

[image]

### Expected results

to see the wiki

### Project file & Debug log uploads

/

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vgdh, 2024-04-18:
wiki desperately needed help of volunteers to fill all the help gaps in the wiki

zimivanende, 2024-04-18:
Unfortunately my knowledge is to limited otherwise I would volunteer. Isnt
there a way to train an open ai model? You put in the code and a
description and it generates it.

closed 2024-07-26 by github-actions[bot]