OrcaSlicer/OrcaSlicer #5067: slice error

windyan9, 2024-04-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

14.0 

### Additional system information

cpu:M1

### Printer

DIY

### How to reproduce

1.create a new project
2.Add a new model(stl)
3.Click "preview"

### Actual results

[image]

### Expected results

no error info

### Project file & Debug log uploads

[cubreboquilla_largo_clarinete-report.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/15020914/cubreboquilla_largo_clarinete-report.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Update from 1.9.x

vgdh, 2024-04-18:
it's not clear what is wrong

windyan9, 2024-04-18:
￼

When appear this message, I can`t continue.
This message appear everytime when i slice a model, I can`t print any model

===============================
杨文锋
QQ：3445599
Mobile/WeChat:18601178314

vgdh, 2024-04-18:
this is a Gcode window. You can disable it by pressing "C" key on english layout keyboard or also in "view:" menu.

mrgfonkyfresh, 2024-04-22:
meme probleme sur macos ventura en francais. on ne peut pas decouper le plateau sans message d erreur. il faut mettre orca en anglais pour que cela fonctionne.
[image]

closed 2024-07-30 by github-actions[bot]