OrcaSlicer/OrcaSlicer #5069: Color Change based on layer still printing waste tower

kugelzucker, 2024-04-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Fedora Workstation

### Printer

Bambulabs X1C + AMS

### How to reproduce

Slice file, right click in sliced view on the layer bar and select color change.

### Actual results

slicer produces a tower (wipe? purge? i dunno) and wastes material doing so.

### Expected results

no tower, just a filament change (AMS) and some poop out the back.

### Project file & Debug log uploads

[bug demo orca.zip](https://github.com/SoftFever/OrcaSlicer/files/15022645/bug.demo.orca.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-04-18:
As far I know it's not a waste it's a prime tower. It's purpose to prime nozzle before printing with new filament and prime some pressure in to the nozzle for printer with bowden systems.

rednaxal, 2024-04-19:
This is not a bug but a feature called Prime tower. You can disable it in the "Others" print settings. But it is better to leave it enabled for a better print quality, just lower the prime volume if necessary.

closed 2024-07-27 by github-actions[bot]