OrcaSlicer/OrcaSlicer #5072: Reducing purge through retracting filament Optimization

boomer20021, 2024-04-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Reducing purge through retracting filament
Optimization has been carried out based on the beta1 feature. A long retraction is performed before the filament is cut, reducing the filament flushing length. This allows for a reduction in flush volume by approximately 2.4*length mm^3.

Now, it's possible to set parameters for individual filaments. These filament-specific parameters will override the parameters of the extruder. If auto-calculaet flush volume when changing filament is enabled, the system will automatically perform the flush calculation when filament is changed. In other cases, please manually trigger automatic flushing.

At present, only the X1 and X1C support this feature on filament parameters, with Bambu PLA Basic and Bambu PETG Basic being default open. Please ensure you're using the latest printer firmware(01.07.03.00). Other printers will gradually get support for this feature. (Note: Currently X1E and P series support this feature on extruder parameters, while the A series do not support this feature.)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Reducing purge through retracting filament
Optimization has been carried out based on the beta1 feature. A long retraction is performed before the filament is cut, reducing the filament flushing length. This allows for a reduction in flush volume by approximately 2.4*length mm^3.

Now, it's possible to set parameters for individual filaments. These filament-specific parameters will override the parameters of the extruder. If auto-calculaet flush volume when changing filament is enabled, the system will automatically perform the flush calculation when filament is changed. In other cases, please manually trigger automatic flushing.

At present, only the X1 and X1C support this feature on filament parameters, with Bambu PLA Basic and Bambu PETG Basic being default open. Please ensure you're using the latest printer firmware(01.07.03.00). Other printers will gradually get support for this feature. (Note: Currently X1E and P series support this feature on extruder parameters, while the A series do not support this feature.)

### Additional context

Reducing purge through retracting filament
Optimization has been carried out based on the beta1 feature. A long retraction is performed before the filament is cut, reducing the filament flushing length. This allows for a reduction in flush volume by approximately 2.4*length mm^3.

Now, it's possible to set parameters for individual filaments. These filament-specific parameters will override the parameters of the extruder. If auto-calculaet flush volume when changing filament is enabled, the system will automatically perform the flush calculation when filament is changed. In other cases, please manually trigger automatic flushing.

At present, only the X1 and X1C support this feature on filament parameters, with Bambu PLA Basic and Bambu PETG Basic being default open. Please ensure you're using the latest printer firmware(01.07.03.00). Other printers will gradually get support for this feature. (Note: Currently X1E and P series support this feature on extruder parameters, while the A series do not support this feature.)

AaronMickDee, 2024-05-06:
I am patiently waiting for this to be added. And by patiently I mean anxiously tapping my fingers on my desk sweating profously.

Foxxytrooper, 2024-05-14:
I am also waiting for this.

In addition: It would be nice to integrate the "prime tower" to an "purge object". At best the slicer should calculate the infill of the "purge object" so that everything of filament change and cleaning is flushed and primed into an object which is roughly only as big/filled as needed.
So every "waste" could be a pencil holder, dice, planter and so on.

dartrax, 2024-07-01:
It's there. Look at #5526, you'll find it under Filament Setting Overrides.

closed 2024-10-07 by github-actions[bot]