OrcaSlicer/OrcaSlicer #5079: Add support back in for multiple extruders/tools (not just multi-filament)

BrewNinja, 2024-04-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If you have multiple extruders or tools, and thus, multiple nozzles sizes, it won't work. It mostly works currently if you have just multi-filament, but even that isn't 100%, as it assumes a single extruder.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Support added back in for multiple separate extruders/tools

### Describe alternatives you've considered

Right now the only alternative is to use Prusa Slicer or Cura

DrKlipper, 2024-04-23:
Great feature request and wanted by a lot of guys. Please take a look at this discussion: 
https://github.com/SoftFever/OrcaSlicer/discussions/931

Having multi extruder support would be an awesome enhancement for Orca. 
And since it is implemented in other Slic3r forks ... It should be doable.

Andy-1664, 2024-04-23:
Hope this gets implemented before I get an xl otherwise I'll be moving to Prusa, I've only got eyes for one slicer.!

Astrocat505, 2024-04-23:
Just finished assembling my ProForge 4 and now I need a slicer other than Cura and Prusa to get my g-codes from. My X-Max 3 and V400 are supported by Orca, now if my ProForge 4 was as well I'd not have to jump back and forth from program to program.

Orca development team, if you all would kindly help us out with this we would be devoted to you and your awesomeness.

sick4noise, 2024-04-23:
I just built a 6-in-1 printer and I would love to be able to use my favorite slicer, Orca. I've been using SuperSlicer in the meantime, but I hate having to use SuperSlicer _just_ for multi-color prints - I find the UI difficult to use.

traxman25, 2024-04-24:
A long overdue upgrade.

diode-ee, 2024-04-24:
This feature would put it above the other slicers. Please implement.

webboty, 2024-04-24:
A bummer that orca is implementing so many tiny features that makes the printing experience better but is missing this major elephant in the room.

I had to go back to prusa slicer as of the missing multi extruder functionality.

BlackShadeOSS, 2024-05-31:
Any updates on this?

avrabi19, 2024-06-07:
is there an update?

akinferno, 2024-06-26:
Says it was added to the Dev Plan on May 9. Still no action taken, unless I am looking in the wrong place. So adding comment to keep this thread fresh. This is becoming a more common feature that is supported by all the other major slicers.  I use Orca for all my other printers, but Prusa for my Stealthchanger.

lukasschwoerer, 2024-06-27:
Multi extruder support would be amazing!! I myself have an E3D Tool changer and would instantly switch to Orca slicer if it would support multiple extruders.

closed 2024-10-03 by github-actions[bot]