OrcaSlicer/OrcaSlicer #5084: Weird internal solid infill pattern on circular models when using monotonic or monotonic line pattern

PEZ18, 2024-04-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

i9-9900k

### Printer

Custom CoreXY

### How to reproduce

Any internal solid infill pattern other than concentric.

### Actual results

weird infill pattern

### Expected results

monotonic like top and bottom surfaces

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I dont know why it is doing this pattern on internal solid infill, but it does this on Monotonic and Monotonic line. Why is top and bottom pattern different from internal solid infill if I set all to Monotonic?

[image]
[image]

Ataraxia-Mechanica, 2024-04-20:
Detect solid narrow internal infill is doing this, disable this would cause prints to go a little slower and your printer to wobble more. Make sure you have your machine's accel to decel set to a reasonable value to not shake your printer apart. (< 70%, usually at 50%)

PEZ18, 2024-04-22:
Thanks! Its been stressing me out and could not figure out what was causing it.

closed 2024-04-22 by PEZ18