OrcaSlicer/OrcaSlicer #5088: Ctrl + Z the application crashes

hliebscher, 2024-04-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-dev

### Operating System (OS)

macOS

### OS Version

14.2.1 (23C71)

### Additional system information

Apple Book Pro M1  16GB 

### Printer

Prusa XL 5T 

### How to reproduce

Whenever I try to undo something and press Ctrl + Z, the application crashes without any error message and closes immediately.

### Actual results

close application

### Expected results

undo last change (it works)

### Project file & Debug log uploads

last thing in log file

[error]	2024-04-20 11:54:27.517970[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.518297[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.518618[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.518981[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.519310[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.519616[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.519929[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.520288[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.520698[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-04-20 11:54:27.521021[Thread 0x00000001de571000]:calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

tested in 3 languages ... the same

netweaver1970, 2024-04-20:
Same here, for some time already. Checked also with latest main build

hliebscher, 2024-04-22:
@SoftFever is fixed with the last PR 5109.     it works now

closed 2024-04-22 by hliebscher