OrcaSlicer/OrcaSlicer #5093: Small area flow compensation causes underextrusion with low layer height

Ataraxia-Mechanica, 2024-04-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Neptune 4 / Voron 0

### How to reproduce

1. Print with default values at 0.2 height and it looks acceptable
2. Run the same settings with 0.1 layer height, total filament extruded estimation in the slicer drops, and underextrusion occurs especially on outer walls
3. Turn off small area flow compensation, no underextrusion occur.

### Actual results

Underextrusion

### Expected results

No underextrusion occurs with small area flow compensation on at low layer heights. Maybe tie the values to line length instead of extrusion length?

### Project file & Debug log uploads

Not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

00raq00, 2024-04-22:
Does it occur on curved or straight walls? If curved check g2/g3 support in your kipper. i had resolution 1.0

Ataraxia-Mechanica, 2024-04-22:
Straight. Mine is at 0.05.

00raq00, 2024-04-22:
> Straight. Mine is at 0.05.

So it is different problem. You are using a more precise value of gcode_arcs than Orca suggests. 👍

jonny7737, 2024-04-30:
On macos, enabling this feature crashes orca on slice. :^)

closed 2024-08-07 by github-actions[bot]