OrcaSlicer/OrcaSlicer #5094: Add 0.15 and 0.25mm nozzle for Bambu Lab

azsenca, 2024-04-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is no 0.15mm and 0.25mm nozzle size for Bambu Lab X1C

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

Please add 0.15mm and 0.25mm nozzle for Bambu Lab X1C, as Biqu has REVO hotend for bambu lab, and REVO has 0.15mm and 0.25mm nozzles.

Eldenroot, 2024-04-20:
You can create your profile

azsenca, 2024-04-20:
> You can create your profile

I know, but it messes then with original 0.2mm nozzle from bambu. Because it inherits 0.2mm nozzle settings and printer, so you have same profiles and filament sunder 0.2mm and 0.25mm printers...

Eldenroot, 2024-04-20:
Create a new profile and printer

azsenca, 2024-04-20:
> Create a new profile and printer

It is not possible to create new bambu lab printer and that it does not inherit filaments and profiles from original bambu lab printer. If it is, please let me know how...

closed 2024-07-27 by github-actions[bot]