OrcaSlicer/OrcaSlicer #5095: Top and Bottom surface pattern to "I don't care, make it fast"

FrAllard, 2024-04-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I always feel frustrated when I notice that the bottom surface requires a significant amount of time to achieve a perfect pattern, especially since I'm not always particularly concerned with the pattern's appearance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would appreciate an option that enables the slicer to render patterns in the fastest way possible, similar to how it processes infill without always adhering to a monotonic line. I suggest prioritizing efficiency and speed above all. For instance, a thin 2 mm wide area around a square could be filled with concentric lines, while other parts could utilize a monotonic line. The goal is to eliminate inefficiencies, such as the printer unnecessarily spending time on filling a 2 mm gap with a monotonic line. This feature would be particularly useful for functional prints where the visibility of the pattern is not a concern.

### Describe alternatives you've considered

I have tried using only concentric patterns on some prints, but it's always a compromise. Certain parts of the model would benefit greatly from concentric lines, while other areas would be better served by monotonic lines.

### Additional context

I would like to see an option near the pattern selection area, featuring a dropdown menu with choices such as 'No Optimization', 'Minimal Optimization', 'Normal Optimization', and 'Maximum Optimization'. This would allow the slicer to use my chosen pattern, such as Monotonic Lines, while also enabling optimization for speed in certain areas by selecting alternative methods. The chosen level of optimization would dictate the slicer's level of creativity. Imagine printing a wheel with spokes: under 'Maximum Optimization', the slicer could use concentric patterns for the circular parts of the wheel and monotonic lines for the spokes, while also varying the printing angles between each section to accelerate the process.

codyglad, 2024-04-24:
Are you suggesting that the printer perform different fill types on the bottom surface depending on the geometry of the part? Ex it would be monotonic on part of the bottom surface and concentric on a different part of the surface?

Would an 'estimated time' in the drop down for the setting be good enough? Or do you want it to be more hands off?

FrAllard, 2024-04-24:
My first tough was to let the slicer use whatever method it want to use to fill the layer the most efficient way, reducing head direction change. Then I tough maybe it would be a good thing if we'd have more control over that feature and giving us a way to control the "main" pattern. But you know what? Having a new pattern called "Efficient fill" would be perfect. That would let the slicer use whatever mean it want to fill using the least amount a direction change to achieve the goal.

As for your suggestion about the "estimated time" I don't understand what you mean. For me a hand off letting the slicer do whatever it wants works for my idea. I would use that for parts where the bottom or top parrent does not matter to speed things up a little bit. Most of the time on the bottom I don't see the pattern anyways, depending of the type of filament I use, I mainly use filament that does not really show the pattern. Maybe my build plate (textured PEI) is better at hiding the pattern...

discip, 2024-07-17:
@FrAllard
As a workaround, you could use **`Archimedean Chords`** for both top and bottom, as this pattern tends to be fastest in almost every case.

FrAllard, 2024-07-18:
Thanks for the suggestion. While this workaround is a viable option, my testing has shown that it can be slower and less efficient compared to using monotonic lines, which provide consistent results. The primary goal of my suggestion is for the slicer to calculate the fastest way to fill the bottom and top layers, prioritizing speed over appearance. Watching my printer fill a 10 mm wide gap with hundreds of narrow passes is quite frustrating sometimes!

closed 2024-10-25 by github-actions[bot]