OrcaSlicer/OrcaSlicer #5100: Orca Tree Supports base / top interface patterns broken/not working at all....and there is no infill on tree support either..

soundtechscott, 2024-04-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 64 bit

### Additional system information

Not really Important for this bug as it happens on multiple machines and is not processor/memory related

### Printer

Neptune 4 Max

### How to reproduce

add any object that would use supports
select tree support - any version - default as an example
choose any base base pattern and any interface pattern

### Actual results

the interface patterns never change away from the rectilinear lines no matter what you set them to
[image]

[image]

there is also no infill in the tree supports and there needs to be
[image]

### Expected results

I expect that if I select Honeycomb, or grid, that it actually gives me the pattern I have selected
I expect that the tree support is not hollow but that it follows the infill the same as the part, or at least would have its own infill parameters that I could use

### Project file & Debug log uploads

I dont have the ability to upload a project file at this time

use literally any file that has an overhang that a tree support would work on

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Jason-Weber-UC, 2024-06-05:
Can confirm.
Creality K1 Max + Orca 2.0.0
Add model. Select Tree support. Slice. Not generating any interface layers.

closed 2024-09-12 by github-actions[bot]

soundtechscott, 2024-09-12:
Still broken...
I've given up on using orca slicer for anything
in Cura, all my prints work with no issues
in orca, I get spaghetti mess with almost everything I print
especially if it needs supports, guaranteed to fail with the supports not working like they should

crookz, 2025-01-02:
Still broken. I have to move to cura also to do anything requiring tree supports that touch the model. Same issue on Bambu slicer with many complaints. No one address this basic issue that has been there a long while yet we get scarf seams. Can we just have moderte functions of supports working. See bambu slicer issue here also: [https://forum.bambulab.com/t/slicer-not-generating-bottom-support-interfaces/36327](url)

crookz, 2025-01-04:
Also related to this issue: [ https://github.com/SoftFever/OrcaSlicer/issues/5357](url)

Milopapa, 2025-02-12:
Facing the same issue. How do you get an issue to reopen here?