OrcaSlicer/OrcaSlicer #5102: X-Y hole compensation issue

gshyl, 2024-04-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0

### Operating System (OS)

Windows

### OS Version

windows11

### Printer

Creatility CR6-SE

### How to reproduce

slice a hollow cylinder with a bridge, with the setting of X-Y hole compensation = 0.2mm
[image]

### Actual results

the thickness of the object changed when the bridge starts being printed

### Expected results

the thickness should be all the same (the object is designed the same), which means that X-Y hole compensation setting should be executed on the whole object. I tried to set the X-Y hole compensation = 0mm, and the problem was solved, but the width of the hole is not correct. it seems that the X-Y hole compensation is only effective on the bridge part.

### Project file & Debug log uploads

[example.zip](https://github.com/SoftFever/OrcaSlicer/files/15051551/example.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DVSVIDEO, 2024-04-23:
I think the "hole" is when the contour returns to itself.
That's how the slicer knows it's a hole.
Where the middle bridge is, that's where the contour breaks. The "hole" there is not a "hole" so it is not compensated for.
I think this is normal behavior. This option was developed for better fitting of narrow holes and not for large diameter round objects.

Fusiable, 2024-04-23:
I've experienced the same issue. Its understandable on how it works, but no matter the combo of Hole compensation to XY compensation is set, there is always a visible "Step" going from a contour to a hole. We need a way to adjust holes without effecting contours on complex parts such as the one above, since undersized holes are usually the main issue (At least for me, on parts that I am not designing)

closed 2024-07-31 by github-actions[bot]