OrcaSlicer/OrcaSlicer #5110: Update to 2.0.0 from 1.9.1 does not keep user settings [still not fixed from previous issue, 1182

The-Monkey-King, 2024-04-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1 Max

### How to reproduce

1. Download version 2.0.0
2. Installation prompts the removal of previous version. 
    NOTE: I made a backup of all my preset filaments, just in case.
3. Installation occurs in the same default locations, no errors
4. Open version 2.0.0
5. Printer defaults to my older Creality Ender-3. I had to go into Printer section and re-select Creality K1 Max 
6. In Filament section, I as presented with only the filament you get when installation of slicer occurs. All of my filaments are missing.
7. Clicking the Filament Gear (Set filament to use), the System Filaments tab defaulted to installation presets with not all filaments selected.
8. Clicking on the Custom filaments tab shows only one filament, not all
9. In the Process section, again only those processes created during installation and not custom defined.
10. [For Windows] I searched in C:\Users\[username]\AppData\Roaming\OrcaSlicer\user\default and found my custom settings intact. The installation of 2.0.0 did not remove them but OrcaSlicer does not import or detect these settings.
11. OrcaSlicer 2.0.0 did keep the list of Recently Opened. 
      NOTE: But if I had a choice, this list would be the very last thing to import. I really want my configurations and settings imported first.
12.  Trying to put the configs and presets back into the system, I see you have to export the settings through OrcaSlicer to create a backup. When you install a new version, there is no prompt to do so. 
   

### Actual results

- The presets and configs are still in C:\Users\[username]\AppData\Roaming\OrcaSlicer\user\default
- The only filament that showed in the Custom tab is the one that appears in: C:\Users\[username\AppData\Roaming\OrcaSlicer\user\default\filament\base

- I need to know how to import these without having to key them all in,

NOTE: Still not fixed from previous versions - see open issue: https://github.com/SoftFever/OrcaSlicer/issues/1182

### Expected results

Instructions should detail how to export settings, configs before installing a new version. Making a backup does not work.
Or, these settings should be detected on installation and kept.
The installation pop-up should warn to make an export, instead it only warns of a downlevel version detected. 
[image]

### Project file & Debug log uploads

[debug_Sun_Apr_14_14_50_18_1404.log.zip](https://github.com/SoftFever/OrcaSlicer/files/15063837/debug_Sun_Apr_14_14_50_18_1404.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

cynfewl, 2024-05-15:
I am having the same issue, but my Bambu printer doesn't seem to be affected.

Also the profiles do exist, because when I try and import them again from the AppData directory I get a prompt asking if I would like to overwrite, but the only process setting available is **Default Setting**.

[image]

jmvaughan2, 2024-06-15:
Problem still exists in 2.1.0-Beta

closed 2024-09-21 by github-actions[bot]