OrcaSlicer/OrcaSlicer #5113: Auxiliary Part Cooling fan not under control

chapterv, 2024-04-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.00

### Operating System (OS)

Windows

### OS Version

Windows 10 LTSC 21H2

### Printer

Bambu lab P1

### How to reproduce

clean install orca, login bambu account sync data
open one user profile filament setting and go to cooling
the auxiliary fan setting is gray

### Actual results

the auxiliary fan setting is gray showing what I have set in the bambu slicer (in my case is 70%)
and when printing aux fan is not automatically turned on, need manual turn on

### Expected results

just like the bambu slicer, the setting box is white, letting me to set the percentage of the fan speed
and when printing the aux fan is started automatically

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15064815/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

chapterv, 2024-04-22:
missed searching for aux fan, similar problems detected

closed 2024-04-22 by chapterv

tmyl8, 2024-06-08:
Same issue