OrcaSlicer/OrcaSlicer #5118: Seam position modifier

foresterr, 2024-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have a model with cylindrical and prismatic sections. I would like to have random seams on the cylindrical parts, and aligned/corner seams on the prismatic ones. I thought the obvious solution would be to use a modifier... but surprisingly, modifiers don't support the "seam position" option. Not sure if the reason for this is somehow fundamental, hopefully not.

There are a number of ideas floating around about enabling more control over seams with modifiers (e.g. seam enforcing/seam blocking via modifier instead of only seam painting), which maybe align with this request; in either case, I don't think this specific feature request was clearly articulated before.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like "seam position" to be an available option for a modifier.

### Describe alternatives you've considered

I have considered using seam painting instead, but with global seam mode set to random, even very tightly painted seams will always be rather weakly aligned ("fuzzy").

DVSVIDEO, 2024-05-01:
I don't think the reason why the seam is not included in the adjustment options is because the layer height is not.
The modifier can also be placed in a way that the body is not applied to the whole layer.
In this case, you cannot set different seam placement and layer height (or anything else) in the same layer.
The solution would be this: 
https://github.com/SoftFever/OrcaSlicer/issues/5225

LE1ED4523, 2024-06-12:
> I don't think the reason why the seam is not included in the adjustment options is because the layer height is not. The modifier can also be placed in a way that the body is not applied to the whole [...]

No you can't choose seam position even you make it two "parts". It can only be set in global.
[image]
[image]
It's acceptable for me if I can do that but I can't.

foresterr, 2024-08-11:
Very good point. I also could achieve what I want if this was a per object setting instead of global only.

foresterr, 2024-11-15:
Bump. Also, I looked at the code for seam generation - looks like it's a whole-model level operation, and modifiers are something below model level, so being able to change seam placement per modifier would be tricky to say the least.

One way that would not require rethinking the whole algo, but would also not be very elegant, would be by pre-emptively generating seams according to all strategies being used anywhere in the current print, then picking from one that is locally desirable. Still, it's way beyond me to try to implement something like this myself.

Azim, 2024-12-21:
Also faced that, such a shame really, wanted to have random seams on fuzzy skin part and aligned seams on the maze part:
[image]

LouisH98, 2025-01-24:
Would just like to +1 the above. Random seam position works really well with fuzzy skin, so it would be great to have it as part of the fuzzy skin modifier.

Dawoox, 2025-03-19:
+1 still not a thing in recent orca version and it's a shame, would really be useful.

Dschuli, 2025-05-17:
At least the height modifier for seam position does make sense to my opinion as I often have parts of models where I dont care and would like to avoid the additional travel and parts where I do. Other modifiers might be tricky as I would assume you have just one seam per layer.....

closed 2026-09-28 by ianalexis