OrcaSlicer/OrcaSlicer #5136: REQUEST: Add USB print support without needing to use GCODE files

R2MGaming, 2024-04-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am new to OrcaSlicer and have tried this out with my current 3D printer using the gcode files with the NEW scarf seam feature, and so far, I like this new slicer for 3D printers, but there was one thing I always had trouble with before I used the Repetier Server for it: "finding an option to print via USB instead of using a gcode file."

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

You know how Ultimaker Cura allows you to print via USB along with Repetier Host? Well there should be an option for all 3D printers to have an option to print via USB without anything else to use.

### Describe alternatives you've considered

1. I've tried updating my Cura software, but there was no scarf seam option.
2. I've tried PrusaSlicer, but I haven't used it for anything which is why I am using OrcaSlicer since it is open source.

### Additional context

Would it be possible for an update to include USB printing in the future?

closed 2024-08-01 by github-actions[bot]

m4rc0d1, 2024-08-04:
Agreed, it would be great if I could add a physical printer connected via USB or serial interface. It would be the same of an ESP3D, but with a direct connection (no ESP no WiFi)

srt-jay, 2025-06-06:
I find it so interesting that the developers are somehow not capable of adding USB and ignore every mention of adding USB support. How is this slicer still even around?