OrcaSlicer/OrcaSlicer #5147: Don't slow down external walls when enforcing minimum layer time

Ataraxia-Mechanica, 2024-04-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Printer slows down, external wall glossiness/ringing pattern/corner shapness changes, making prints ugly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Slow down internal parts only and don't slow down outer walls.

igiannakas, 2024-04-25:
That's an interesting request - basically what you're saying is reduce internal features to the minimum print speed before touching the external features, right?

igiannakas, 2024-04-25:
Try this build here:https://github.com/SoftFever/OrcaSlicer/pull/5148

Seeking testers and feedback for this before I invest any more time in polishing this and making it a feature.

Ataraxia-Mechanica, 2024-04-25:
Yep, printing inner features slower feels weird but it makes sense when trying to make outer surface more consistent. I'm going to test this the moment the build completes.

Ataraxia-Mechanica, 2024-04-25:
I don't think it's working for me. In some layers, travel speed is applied to the outer wall while on some it's 10mm/s.
[image]
[image]

Ataraxia-Mechanica, 2024-04-25:
This is the same model, in 2.0. 
[image]

igiannakas, 2024-04-25:
Fixed: 
[image]

igiannakas, 2024-04-25:
Latest build is here: https://github.com/SoftFever/OrcaSlicer/actions/runs/8832248258

Ataraxia-Mechanica, 2024-04-25:
Weird. Looks like if dropping the inner lines' speed doesn't slow it down enough, it immediately sets the outer wall's speed to a very low value instead of gradually slowing it down. Or was this feature a special case and not implemented yet?
[image]

igiannakas, 2024-04-25:
> Weird. Looks like if dropping the inner lines' speed doesn't slow it down enough, it immediately sets the outer wall's speed to a very low value instead of gradually slowing it down. Or is this feature a special case and isn't implemented yet? ![image](https://private-user-images.githubusercontent.com/111327195/325642863-d668f001-d09f-4004-94b4-4d72c02ce66e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwNTg5MDgsIm5iZiI6MTcxNDA1ODYwOCwicGF0aCI6Ii8xMTEzMjcxOTUvMzI1NjQyODYzLWQ2NjhmMDAxLWQwOWYtNDAwNC05NGI0LTRkNzJjMDJjZTY2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNVQxNTIzMjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYTk4NjE3NjIzYjlkMjU1MjI1YjY4NjIyM2VkZWVlOGU4MjE0OWJkMjczMGQ4NjgxMGZlMjY1MTk1ZDRkMDM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lIz4SycO-e3RG5vTAjWcLnNPrWIHI9HC_AMw8T4-nYk)

Drop me your project file as a zip file here to take a look.

also try slicing with slowdown for overhangs disabled - there maybe some interference there

Ataraxia-Mechanica, 2024-04-26:
Yeah, even with class mode it's acting weird. Currently thinking about what to print that has no overhangs and is small enough that the artifact due to slower outer wall would be obvious.

igiannakas, 2024-04-26:
> Yeah, even with classic mode it's acting weird. Currently thinking about what to print that has no overhangs and is small enough that the artifacts due to inconsistent outer wall speed would be obvious.

A benchy works well ;)

closed 2024-06-16 by SoftFever