OrcaSlicer/OrcaSlicer #5152: Multi color painting only applies to top/bottom shell layers

cerrunos87, 2024-04-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz, 2592 Mhz, 6 Core(s), 12 Logical Processor(s)
Memory: 16GB Physical, 27.8 Virtual
GPU: Nvidia RTX 2060

### Printer

Voron 2.4

### How to reproduce

1. Import Test Object
2. Paint surfaces for different filament colors (Circle, Sphere, and/or Fill)
3. Slice

### Actual results

Object is sliced with filament changes only applied to top and bottom shell layers, not vertical shells.
[image]
[image]

### Expected results

Objects should be sliced with surface colors matching painting performed.
[image]
[image]

### Project file & Debug log uploads

[MMU Malenia bug.zip](https://github.com/SoftFever/OrcaSlicer/files/15118274/MMU.Malenia.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

BioCookieYT, 2024-04-27:
Experiencing the same behavior, hope a fix is on its way soon

Coop25, 2024-05-13:
I am experiencing something similar as well, it will start off with the correct color and the 2nd to the top layer it will switch to the wrong color, fill in the object and then switch back to the correct color.

3rd layer from top
[image]

2nd layer from top
[image]

top layer
[image]

I can also provide my project file if needed

raczak, 2024-06-04:
Hi, same issue here. It's changing filament only on top shell surface since the 2.0.0 OracaSlicer version. On 1.9.X it's working great... Don't know why ...
[image]

st3venb, 2024-06-11:
Experiencing the same issue here with 2.1.0-BETA.  Opened up Bambu Studio loaded project file and verified painting works on vertical surfaces there.  This is frustrating to have to use two slicers to achieve my end goal... Subscribed for hopefully a fix later.

atl285, 2024-06-14:
I have the same experience. Color painting is not useable at the moment with Orca Slicer (Version 2.0 & 2.1.0 Beta). I tried to print [this model](https://www.printables.com/de/model/633452-bull-flexi-keychain/files) in 2 colors using the .3mf file.
[image]

@SoftFever, did you know it?

SuddenlyPineapple, 2024-06-14:
Same: 
[image]
[image]

ExR90, 2024-06-22:
I am having this exact problem. The linked fix in PR 5774, re: Filament shrink, for me is already at 100 so I am not sure that fix is related to this problem.

igiannakas, 2024-06-22:
Can you post your project file? It helps with trouble shooting as I cannot reproduce this issue right now.

SuddenlyPineapple, 2024-07-01:
It was fixed in https://github.com/SoftFever/OrcaSlicer/pull/5774 . Install version 2.1.1 ;) I think this issue might be closed :)

closed 2024-07-01 by SoftFever