OrcaSlicer/OrcaSlicer #5153: "Fill bed with copies" should consider the print sequence

tlhintoq, 2024-04-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There's no point putting 100 objects on a bed when the print sequence is already set to "by object" and thus only 5-10 can fit.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Take the 'by object' spacing/logic into account when "fill bed" is executed.

### Describe alternatives you've considered

Fill the bed with the existing code
Then auto arrange that bed only since that spaces correctly.
Deleted all the newly manufactured plates and objects so you are left back at the original plate.

### Additional context

[image]

tlhintoq, 2024-05-10:
Bot nudge.

tlhintoq, 2024-08-09:
"Bot don't close me know" - This is still very much needed.

theherox, 2024-08-14:
The desired functionality described by @tlhintoq was working for me in Orca 2.0.0.  I just updated to 2.1.1 last week and have been having the same issue ever since.

tlhintoq, 2024-11-16:
Good to know it wasn't doing this in 2.0 - thanks @theherox 
So that means it shouldn't be too hard to see what was changed between then and now.  Right?

PR #8038 (2025-01-22): Fix arrange when print-by-object and have object off-plate
Fix #5153
Fix #7745

Partly revert 0b6a1d36360448c5e6818001b9d8ea99a57567a7
closed 2025-01-22 by Noisyfox