OrcaSlicer/OrcaSlicer #5154: Flashforge is violating OrcaSlicer's GPLv3 license.

nativeit, 2024-04-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.1.0-dev

### Operating System (OS)

Linux, macOS, Windows

### OS Version

All

### Printer

FlashForge Adventurer 5M/Pro

### How to reproduce

1. Go to Flashforge's website
2. Navigate to Support > Downloads
3. Download "Orca-Flashforge" for either MacOS or Windows
4. Notice their own restrictive commercial license packaged with this release
5. Try to find the published source code (hint: the fork on their official Github is *not* related to the packaged binaries on their website).

### Actual results

> 2. End User Software Licence
> 
> You are granted a limited and non-exclusive licence to access and use the Software (as defined in Section 1.1) on only one Flashforge device or computer, subject to, and in accordance with the terms and conditions of this Agreement. The Software cannot be sold, transferred, or used for any other commercial purposes.
---
>3. Your Obligations
>
>By using the Product and agreeing to this Agreement, you assure:
>
>3.1 You may not use Flashforge technology or Flashforge intellectual property to develop software or design, develop, manufacture, sell, or licence third-party devices/accessories associated with Flashforge Product without Flashforge's prior consent.
>
>3.2 You may make a copy of the Software on a computer hard disk drive or other storage media for the purpose of archiving. You may not distribute or provide the Software through networks so that it is available to multiple devices or computers at the same time.
> 
>3.3 You may not sell, rent, lease, lend, sublicence or distribute the entirety or part of the Software to any third party without prior written permission from Flashforge. Nonetheless, you may make a one-off permanent transfer of the entire licence to the Software to another party along with the transfer of ownership of your Flashforge device, provided that: (1) the transfer must include your Flashforge device and all the Flashforge Software; (2) you do not retain all or part of the Software copies; (3) the party that receives the Flashforge Product has read and accepted all the terms and conditions of this Agreement.
> 
>3.4 Except as otherwise expressly permitted, you shall not, nor allow any other person to abuse, intrude or make other inappropriate use of the Product, including but not limited to modifying, decoding, copying, disassembling, publishing, publicly disseminating, decompiling, reverse engineering, exporting codes, or creating derivatives of the Product in any way.

### Expected results

Literally _anything_ else.

### Project file & Debug log uploads

[English.zip](https://github.com/SoftFever/OrcaSlicer/files/15125520/English.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

License file from Orca-Flashforge.pkg > Contents > Resources > English.lproj

nativeit, 2024-04-26:
I received the following response after reaching out to their technical support seeking information on the source code for the "Orca-Flashforge" binaries they are distributing on their website:

> On 2024-04-26 01:00, FFsupport wrote:
> 
> Dear
> Sorry for the late reply.
> 
> We have consulted our engineer,we don't have the Orca-Flashforge
> source code.
> 
> Wish you a happy life.
> 
> 点击满意度评价 [[1](http://servicego.udesk.cn/site/public/dynamic-form/index/7f6be7e3-23b2-45c6-ab56-fd0fa19a773d)]
> 
> Best Regards,
> 
> Lisa（邵莉沙）|Aftersales

nativeit, 2024-04-26:
I received an email from their support overnight, they have posted the source for Orca-Flashforge on their GitHub: https://github.com/FlashForge/Orca-Flashforge

I have not had an opportunity to audit this repository, but it's the first step in the right direction anyway. I will run a diff later today and post back with any notable findings.

Avpman2, 2024-04-28:
Hi, Any findings yet on your diff?
Thanks!

SoftFever, 2024-05-01:
Thank you @nativeit 
I will take some time to read through it first.
I really appreciate your efforts to defend open source!

Avpman2, 2024-05-01:
Ditto +100

vgdh, 2024-05-02:
❤️

keithel, 2024-05-09:
Re: "Orca" being used in the name of their forked product - that is governed by trademark law. If Orcaslicer has been trademarked, if suit was brought, a lawyer could argue, and a judge could rule that use of "Orca" violates the trademark (even though the trademark in my example isn't over "Orca" but "Orcaslicer"), but I'm not sure how likely. Better would be if "Orca" was trademarked for use in 3d printing areas - seems likely that both "Orca" and "Orcaslicer" could be trademarked with the narrow focus of additive manufacturing or maybe even manufacturing as a whole.

Re: not making "official forks" -- this isn't something that the GPLv3 or any other OSS license (that I am aware of) mandates, so this demand/request is just a nicety that it would be good if Flashforge were to do. Makes the job of tracking down differences much easier, but not required for them to comply.

@nativeit Thank you for spending your free time working on this very important issue. For the consumer 3D printing industry to succeed as it has for the past 15+ years, we need to make sure that companies are held to account for their use of all the hard work that people have put in over the years.

kylek29, 2024-06-08:
> @SoftFever have you trademarked "Orca" and/or "Orcaslicer" anywhere? I think maybe that should happen if it has not already.

IANAL, but this gets a little tricky with software because of the international nature of it, but in the US you can trademark passively, known as a "common law" trademark (normally indicated by putting ™ on the branding/slogans) -- this signifies that you intend to defend and are laying claim to the use of. If you officially register the trademark, you'd want to put the ® (for registered). The latter gets you more protection.

https://digital.gov/resources/u-s-trademark-law/

nativeit, 2024-06-20:
I really appreciate the attention on this! I work I.T. with a professional baseball team, and have been quite busy the last few weeks, but I have not forgotten about it, and will continue to pursue some sort of mutually beneficial outcome that helps everyone involved here, but especially the users of OrcaSlicer's incredible software, and FlashForge's otherwise solid intro hardware. 

My current position is that there doesn't appear to be any good reason to fork OrcaSlicer at all, and the more open and convenient approach would be to integrate FlashForge hardware as fully into the OG OrcaSlicer as best as possible. Probably won't be super popular with the suits in China, but I will talk to the folks who have already done a lot with the firmware, and see if there's some way to bring next-level FlashForge support to OrcaSlicer unilaterally.

Keep up the great work, @SoftFever and team!

nativeit, 2024-06-21:
*One of many issues with companies co-opting open source software like this!*

A few quick notes about the image below:

1. They haven't even bothered to change the developer on the MacOS binaries. 
2. It's highly unlikely such a lazy effort will receive any meaningful support. 
3. It's unreasonable to expect @SoftFever to support this, as they have no way of knowing what's been tampered with.
4. Because of the aforementioned laziness of this unlicensed clone, the original OrcaSlicer name and branding is all over the place. So if their releases are buggy, no matter what the actual source of the issues are, it _will reflect badly_ on OrcaSlicer and negatively impact their reputation. Something that, from my personal perspective, is currently about as good as one could ever hope for on a FOSS project such as this.

OrcaSlicer deserves all the success and good reputation they have earned, and are still coming to them! Usurping their good name for a half-baked, vendor lock-in, stripped down, objectively worse version of the original is a really pathetic move, and as much as I want to print directly to my networked Adventurer 5M, I will hack its firmware before I support "OrcaFlashforge". I will document the effort, I will donate to the software projects involved, and I will publish tutorials for future Adventurer 5M owners to do the same, complete with full credit and donation links to the people that actually committed the hard work necessary for OrcaSlicer, Klipper3D, and the reverse engineering effort for the Adventurer 5M firmware.

So, anyway, tldr; @FlashforgeOfficial - your sofware is crashing on MacOS:

[image]

SoftFever, 2024-06-30:
> Re: "Orca" being used in the name of their forked product - that is governed by trademark law. If Orcaslicer has been trademarked, if suit was brought, a lawyer could argue, and a judge could rule that use of "Orca" violates the trademark (even though the trademark in my example isn't over "Orca" but "Orcaslicer"), but I'm not sure how likely. Better would be if "Orca" was trademarked for use in 3d printing areas - seems likely that both "Orca" and "Orcaslicer" could be trademarked with the narrow focus of additive manufacturing or maybe even manufacturing as a whole.
> 
> Re: not making "official forks" -- this isn't something that the GPLv3 or any other OSS license (that I am aware of) mandates, so this demand/request is just a nicety that it would be good if Flashforge were to do. Makes the job of tracking down differences much easier, but not required for them to comply.
> 
> @nativeit Thank you for spending your free time working on this very important issue. For the consumer 3D printing industry to succeed as it has for the past 15+ years, we need to make sure that companies are held to account for their use of all the hard work that people have put in over the years.
> 
> @SoftFever have you trademarked "Orca" and/or "Orcaslicer" anywhere? I think maybe that should happen if it has not already.

Nah, it doesn't.
As long as one follows the GPL license and clearly credits OrcaSlicer, I have no problem (in fact, I'm happy) with anyone forking Orca.
This is what Open Source is meant to be.

However, I also agree with what @nativeit said: if a company isn't committed to investing heavily in a slicer (which is a complex and large project), it's probably not a good idea to fork Orca just for the sake of "owning" one. It’s not fair to their users, as maintaining software requires a lot of skill, knowledge, and dedication.

keithel, 2024-07-01:
@SoftFever I think maybe you misunderstood me. I also don't have a problem with folks forking GPL projects and adding their own modifications, presuming they follow the terms of the GPL, including crediting the prior authors.

What I am suggesting though - is making sure there are some protections over the name "OrcaSlicer" as it is applied to something in the additive manufacturing world. Fork, yes, but use the name "OrcaSlicer" - that's where I personally think a company has gone too far. I'm certainly not speaking for you, so if you feel differently then by all means, that's how things should be treated.

SoftFever, 2024-07-09:
> @SoftFever I think maybe you misunderstood me. I also don't have a problem with folks forking GPL projects and adding their own modifications, presuming they follow the terms of the GPL, including [...]

Thanks @keithel 
I totally agree with you here.

nativeit, 2024-07-21:
I had a friend send me this earlier, apparently this thread got a quick screen cap on a YouTube video[1] posted a few days ago on the channel [@thenextlayer](https://www.youtube.com/@thenextlayer). 

Considering it's a lot of inside baseball on some fairly niche topics, I'd say it's performing rather well, getting to nearly 50k views in the last two days.

I am not familiar with the person who made the video, but I assume they're sincere, and they have a lot good things to say. It's focused on the 3D printing industry in particular, but a lot of the points they raise are the same kinds of issues--very legitimate and ongoing issues--that the FOSS community have been working to confront. I have already filed a complaint with the Free Software Foundation (GNU)[2] and I encourage everyone reading this to take a few minutes to do the same.

Whatever else happens with this, I just want to underline that OrcaSlicer is a **_f'cking impressive accomplishment_**, and it's made my own experiences with this new hobby _way_ more enjoyable and productive. Thank you, _so_ much.

If anyone's got an extra few bucks, please consider sponsoring @SoftFever and OrcaSlicer to support their ongoing development efforts.[3]

### Links:

1. **An Open Letter to the 3D Printing Community** [video]: [https://youtu.be/YRayFmMYFxM](https://youtu.be/YRayFmMYFxM) 
    - This thread appears at [4:49](https://youtu.be/YRayFmMYFxM?t=236)
2. **GNU License Violations Reports**: [https://www.gnu.org/licenses/gpl-violation.en.html](https://www.gnu.org/licenses/gpl-violation.en.html)
3. **Donate to OrcaSlicer** - [https://github.com/sponsors/SoftFever](https://github.com/sponsors/SoftFever)

AJolly, 2024-08-06:
I wish they would publish proper sources for Klipper on the 5m, they are completely out of compliance with that as well :/

kylek29, 2024-08-09:
I noticed the latest Creality Slicer is based on OrcaSlicer, has anyone more knowledgeable on these licenses checked them for compliance?

https://www.creality.com/pages/download-software

It's getting silly how many of these companies are forking just to add their logo and a few features. Why not just sponsor the project and add features to the main?

closed 2024-11-15 by github-actions[bot]