OrcaSlicer/OrcaSlicer #5156: Model is sliced with huge gaps

Rickthebig, 2024-04-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

DIY BLV Cube based on Troodon (Reprap) as template

### How to reproduce

Slice following model:
[image]

### Actual results

Result of slicing:
[image]

Note the gaps in the walls.

### Expected results

The model should be sliced without gaps.

### Project file & Debug log uploads

[Uploading Chev_Express.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-04-26:
May be STL is broken, try use repair tool.

Rickthebig, 2024-04-26:
> May be STL is broken, try use repair tool.

Does Orca have a repair tool? If not, which one would you recommend?

buzzhuzz, 2024-04-26:
Orca does have model repair tool in its Windows build. Unfortunately, I can't help much there as I'm only use Linux.

Also, sometimes exporting STL back from Orca into the file and re-import it again helped me with slightly broken models (like a dozen of open edges).

Rickthebig, 2024-04-29:
I can confirm that the model is broken. This is not a bug in Orca.

closed 2024-04-29 by Rickthebig