OrcaSlicer/OrcaSlicer #5167: Crashing when initiating an export of configs

Gozania, 2024-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Home Version 23H2

### Additional system information

Device name	Legion-5i-Pro
Processor	11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz   2.30 GHz
Installed RAM	64.0 GB (63.8 GB usable)
Device ID	4CDC760F-0F15-42F2-96EA-3C074504F143
Product ID	00342-20727-75696-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

None (Splash screen) but presists no matter which printer selected

### How to reproduce

1. Open Orca Slicer
2. Click "File"
3. Hover over "Export"
4. Click "Export Configs..."
5. Program Crashes.

### Actual results

Program terminates itself.

### Expected results

A window should pop up to select directory & file name of the configs to be exported.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15137256/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RoisinDubh79, 2024-05-13:
I'm seeing this issue as well when trying to export my config.

Gozania, 2024-05-13:
> I'm seeing this issue as well when trying to export my config. Issue also happens with the nightly build of 2.1 Dev

Hope its something that can be ironed out in the next RC.  Would be invaluable when switching machines or setting up orca on several machines.

RoisinDubh79, 2024-06-01:
Still crashing with the 2.1 beta

Gozania, 2024-06-23:
Still crashing after updating to 2.1.0 full release.  We need this far more than a pretty new splash screen...  Will someone please pull this and get it taken care of?

RoisinDubh79, 2024-06-27:
It seems this could be related to memory.
My laptop has 16 GB of RAM and it still fails for me.
Someone I know uses Orca with a computer that has 32 GB of RAM and does not encounter the error.

Gozania, 2024-06-28:
> It seems this could be related to memory. My laptop has 16 GB of RAM and it still fails for me. Someone I know uses Orca with a computer that has 32 GB of RAM and does not encounter the error.

Im running 64gb.... Not the issue.....

[image]

Paul-Damian, 2024-09-08:
Same with the latest v2.2.0-dev

closed 2024-12-15 by github-actions[bot]