OrcaSlicer/OrcaSlicer #5168: All user settings dissapear after signing in

Gozania, 2024-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Home Version 23H2

### Additional system information

Device name	Legion-5i-Pro
Processor	11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz   2.30 GHz
Installed RAM	64.0 GB (63.8 GB usable)
Device ID	4CDC760F-0F15-42F2-96EA-3C074504F143
Product ID	00342-20727-75696-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Ender 3 S1 Plus/Ender 3 V2/Bambu X1C

### How to reproduce

1. Open Orca Slicer
2. Go to Prepare
3. All User Printers/Filaments/Profiles all present
4. Sign In
5. Go to Prepare
6. All User Settings are gone.
As soon as I sign in all my personal settings totally vanish from pulldowns.

### Actual results

[Orca.zip](https://github.com/SoftFever/OrcaSlicer/files/15137306/Orca.zip)

### Expected results

All my settings should remain when I log in.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15137329/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I did import all my settings via the perl script found here.  https://github.com/theophile/SuperSlicer_to_Orca_scripts   I would just export and import my settings when logged in, however I am unable to export  https://github.com/SoftFever/OrcaSlicer/issues/5167 

I imported all my settings prior to creating a bambu studio account.  Also when I do log in I select NO when it asks if I want to sync settings from bambu cloud.  (I have yet to use it since I am about to purchase my first bambu machine.

Gozania, 2024-04-29:
I achieved what I wanted, please close.

TexZeTech, 2024-04-30:
You can just hit "close issue" .

closed 2024-05-01 by Gozania