OrcaSlicer/OrcaSlicer #5170: No video from X1C or P1S, yet Bambu Slicer no problem.

tlhintoq, 2024-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 nightly since 25apr2024

### Operating System (OS)

Windows

### OS Version

Windows10x64

### Printer

Bambu X1C or P1S

### How to reproduce

From Device tab, try to see streaming video from either printer

### Actual results

[image]

As you can see, Bambu slicer can connect.  
Log out then close down Bambu slicer.  Start Orca then log in to Bambu network.  Nope, no video - but will print to it just fine.

### Expected results

If Bambu Slicer and see the video feeds then so should Orca Slicer, right?

### Project file & Debug log uploads

Let me know what you need.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Maybe a Bambu network plugin that didn't update with the release version of Orca 2?
What do you need me to look at to confirm versions?

KD7LRJ, 2024-04-28:
Same for me. Handy works on my Android phone, but 2.1.0-dev gives a black screen with an error message, "Please check the network and try again. You can restart or update the printer if the issue persists. [2:-10]"

Same results for my 4 P1P and 2 X1C machines.

Un/reinstall hasn't helped. Send/Printing from my Win 11 desktop works fine, just not the video feed.

tlhintoq, 2024-04-28:
Hey Douglas, Thanks for the confirmation that it wasn't just me - I mean I'm sorry its happening to you, but its good to know its more widespread than just one bad installation.

73's
KC9RY - Clint

tlhintoq, 2024-04-29:
Just downloaded 2.1dev 28apr.
Uninstalled old version.
Ran installer as admin.
No flags, errors, warnings, issues.

Ran the newly installed Orca.  I was hopeful when I saw it downloading the Bambu network plug-in, thinking that might be latest everything.
[image]

No joy.  Still able to print via network to X1C and P1S - but no video stream from either.
FYI: I have a static IP to my house, so its not NAT or some other broadband issue.
[image]

TexZeTech, 2024-04-30:
[image]

@tlhintoq You didn't upload any log files or project files...... kinda hard to look at what may or may not be happening with out those. So to start upload your log files.

tlhintoq, 2024-05-01:
[image]
[debug_network_Wed_May_01_13_44_16.log.zip](https://github.com/SoftFever/OrcaSlicer/files/15173230/debug_network_Wed_May_01_13_44_16.log.zip)

pgifford, 2024-05-01:
This should be easily reproducible.  The error code (from [here](https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view)) is: 

> <html><body>

-10 | Certificate error | Check if you are using the official version of Bambu Studio
-- | -- | --

</body>
</html>

As noted Live View works fine in Bambu Studio, also works in Bambu Handy.

tlhintoq, 2024-05-09:
X1C got latest firmware yesterday.
Today OrcaSlicer just prompted for new network plug in.  
This seemed like a match and maybe would resolve the issue.
No joy.  Still fails with the `2:-10` error

tlhintoq, 2024-05-10:
10may2024
Grabbed the latest overnight.
Ran installer as admin
Let it uninstall previous version.
Installed new version.
Run it
Prompt for new network plugin - `[OK]`
Close OrcaSlicer (because it takes effect on the next launch)
[image]

Launch again.
Log out of Bambu network
Log in to Bambu network
power cycle the X1C and the P1S

All other network functions are good.  But no video stream.  BambuStudio has no problem with video stream so its not the network, PC or printers.
[image]

tlhintoq, 2024-05-12:
New network plugin alert.
[image]
Same proper steps as before including reboots etc.
No joy.  Still getting no video with `2:-10` error.

pgifford, 2024-05-12:
Video is working for me now after the latest network plugin installed.

tlhintoq, 2024-05-13:
> Video is working for me now after the latest network plugin installed.

What is "latest" for you?  "01.09.01.06" like in my screen shot, or newer?
Which firmware on your machine?  X1C or P1S ?
Which Orca?  2.1.0-dev?  Nightly build?

I'm running dev from 2 days ago.
[image]

I just opened BambuStudio to deliberatley log in then log out.  So I'm not connected there.
Launched OrcaSlicer - log in then log out.  Not connected there.
Powered off both printers.
Powered off network router.
Basically rebooting the entire shop.
Power on the router
Powered on the printers.
Launch Orcaslicer - still get `2:-10` error on both X1C and P1S.
Close Orca, launch Bambu - video on both.
Close Bambu, launch Orca - both still fail.

[image]

[image]

pgifford, 2024-05-13:
I'm using Orca Slicer v2.1.0-dev from 30 Apr 2024.  I don't know how to find the version of the network plugin being used.  When the "new network plugin" dialog pops up I always click OK. 

I just downloaded and ran the latest nightly Windows portable, video works fine.

A few days ago I installed Bambu Studio Beta for an unrelated reason, I don't know how or why that would affect Orca Slicer though but the timing is about right.  One nice thing about Bambu Studio is the Help | About box shows the plugin version.

[image]

tlhintoq, 2024-05-13:
>  just downloaded and ran the latest nightly Windows portable, video works fine.

Hmmm... "Portable version", huh?  I might have to try that and see if there is a difference to the full installed version.

tlhintoq, 2024-05-13:
>  I might have to try that and see if there is a difference to the full installed version.

No joy.  Same error.

pgifford, 2024-05-13:
> > I might have to try that and see if there is a difference to the full installed version.
> 
> No joy. Same error.

Try installing Bambu Studio Beta. I can’t say it fixed the problem but I can’t say it didn’t either.

tlhintoq, 2024-05-13:
> > > I might have to try that and see if there is a difference to the full installed version.
> > 
> > 
> > No joy. Same error.
> 
> Try installing Bambu Studio Beta. I can’t say it fixed the problem but I can’t say it didn’t either.

What the heck...
Bambu beta installed.  It grabbed a network plug in.  It sees my printers.
[image]

Now to Orca
[image]

Holy crow - @pgifford you're a life saver.  Looks like the plug in from the beta is the key.  Must be a shared component between both applications.  Keeping in mind, updates to Orca could overwrite it.  _MIGHT_ have to re-install Bambu beta after any Orca updates if the video issue returns.

KD7LRJ, 2024-05-13:
Confirmed! I hadn't updated Bambu Studio since Version 1.8 and the update process installed 1.9.1.66 (not beta). Once I started the app after installing, it also updated the network plugin. Orca Slicer is now working again to view video from my printers.

mydevpeeps, 2024-06-03:
confirmed as well. worked fine on 2.1.0.

pwr22, 2024-06-15:
I had the same issue after updating to Orca 2.1.0 beta and resolved it now by letting Bambu update. Thanks for the help folks 👍!

brandonkish, 2024-06-19:
Using Orca Slicer 2.1.0 kept getting error "[2-10]" in Orca Slicer on 6/19/24 and before.

Updated to Orca Slicer 2.1.0-rc on 6/19/24. Continued to get error "[2-10]"

Confirmed currently installed Bambu Lab version is V1.8.2.56. Can see the camera's video in Bambu Lab V1.8.2.56. 

My Log files with Bambu Lab to V1.8.2.56 installed and Orca Studio V2.1.0-rc installed. With Orca Studio V2.1.0-rc still giving "[2-10]" error.
[log.zip](https://github.com/user-attachments/files/15895504/log.zip)

Downloaded and installed Bambu Lab 01.09.02.57.

Opened Bambu Lab V01.09.02.57. Confirmed I can see the camera video. Closed Bambu Lab V01.09.02.57. (Note: No notification that the "Network Plugin" need to be updated with this version after install)

Opened Orca Slicer V2.1.0-rc. Can now see Video on Device tab.

This is the logs for Orca Slicer V2.1.0-rc after downloading Bambu Lab V1.09.02.57 and being able to successfully see the video on the device tab.
[Orca Slicer Log After Installing Bambu Lab V1.09.02.57.zip](https://github.com/user-attachments/files/15895593/Orca.Slicer.Log.After.Installing.Bambu.Lab.V1.09.02.57.zip)

This is probably unrelated, but I got this warning message constantly when switching between the programs and after updating each program:
[image]
In bot Orca Slicer and Bambu Lab when I updated Orca to 2.1.0-rc, and Bambu Lab to V1.8.2.56. Not sure if this is related, but I hit "Yes" both times. Which brought up this window:
[image]
Clicking "Yes" to this prompt did not seem to allow the video to display in Orca before updating Bambu Lab.

brandonkish, 2024-06-19:
I also should note, at one point, when I installed Orca Slicer 2.1.0, Orca slicer gave me some kind of message that it could not load my printer configurations. This happened a month ago or so. The exact date is not clear to me.

After a brief moment of panicking, I restarted Orca Slicer, and all my profiles were sill there. I don't remember what I did to bring them back, or if I did anything. I am not sure if this is related, and which I could give you more details, but I am not sure if this had something to do with it, or if others also had this issue.

mydevpeeps, 2024-07-04:
still an issue in GA 2.1.1 see https://github.com/SoftFever/OrcaSlicer/issues/5754#issuecomment-2209553656 for workaround.

drizztguen77, 2024-07-08:
I was receiving the exact same warning about BambuSource from a different install. I tried the 2.10 version and then the 2.0 version. The 2.10 version still received the warning and then 2.0 version works fine. So there is something new in the 2.10 version that is causing an issue with the network driver. If I say "No" to not fix it, it still seems to work and shows me the camera.

mlimseth, 2024-08-20:
Just install latest Bambu studio and the video come back on Orca.

drizztguen77, 2024-08-20:
I have the latest version of Bambu Studio installed. I installed Orca 2.1.1 alongside 2.0.0. I get no warning at all when I use 2.0.0. When I use 2.1.1 I get the warning. This is with Bambu Studio 1.9.3.50 installed. I can answer No and it seems to work but it is still concerning to get the warning.

[image]
[image]

closed 2024-11-26 by github-actions[bot]