OrcaSlicer/OrcaSlicer #5171: multiple GTK errors on KDE wayland

Nathan22211, 2024-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Arch Linux (zen kernel)

### Additional system information

ryzen 7 5700X
rx 7600

### Printer

can't even get it to launch

### How to reproduce

open the app image under KDE wayland on an arch linux install. mind you that webkit2gtk and qt5-wayland are needed on the system to fix some other errors

### Actual results

it fails to open

### Expected results

it should open as expected

### Project file & Debug log uploads

I only really have the console output to go off of
```
[2024-04-27 11:41:02.195490] [0x00007604c63f9480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.547: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.547: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.547: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.547: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.548: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.548: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.548: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.548: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.549: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.549: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.550: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.550: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.550: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.550: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.551: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.551: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.551: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.551: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): GLib-GObject-CRITICAL **: 11:41:02.552: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.552: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.649: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.651: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.656: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.656: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.656: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.719: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.719: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.727: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.727: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.729: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.729: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.734: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.734: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.738: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.738: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.742: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.742: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.746: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.746: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.750: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:286265): Gtk-CRITICAL **: 11:41:02.750: gtk_widget_set_size_request: assertion 'width >= -1' failed
```

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I don't think it's related to a KDE sizing issue. mine was at 100% and even tried it at 75% to the same effect

patrickslarson8, 2024-05-13:
Experiencing the same issue on a fresh install of Ubuntu 22.04, so I don't think it's related to the KDE environment.

patrickslarson8, 2024-05-13:
Seems to be related to the Bambu Studio code that was pulled in. See also: #886 and #https://github.com/bambulab/BambuStudio/issues/1868.

spusuf, 2024-06-27:
Same issue with basically identical logs on Arch Linux using Wayland.

serchemach, 2024-06-28:
Had the same issue with Arch Linux using hyprland, was able to solve it by using setting GDK_BACKEND=x11 when running the appimage

trebory6, 2024-08-15:
> Had the same issue with Arch Linux using hyprland, was able to solve it by using setting GDK_BACKEND=x11 when running the appimage

Can you elaborate how you did that?

serchemach, 2024-08-16:
> > Had the same issue with Arch Linux using hyprland, was able to solve it by using setting GDK_BACKEND=x11 when running the appimage
> 
> Can you elaborate how you did that?

It seems I had some issues with xwayland, so I had to manually force X11 for orca by launching it like this: "GDK_BACKEND=x11 OrcaSlicer"
But after a few updates this stopped happening to me for some reason and I could just launch it normally

jordicunillerarivera, 2024-08-17:
Same here, i tried AUR and appimage versions, with and without GDK_BACKEND=x11, all the dependencies up to date, none of that fix the errors, the app seems to work fine but the dropdown menus didn't open, so i only can print by defaullts options.

- Hyprland Arch Linux

Nathan22211, 2024-08-17:
For me only 2 depencies were needed manually for the AUR version for it to
work, but that was a few updates ago

Nathan22211, 2024-08-17:
> You remember what dependencies was?

Webkit2gtk, gst-plugins-base, and gst-libav. The last one is only needed
for the X1C camera

jordicunillerarivera, 2024-08-17:
> […](#)

Sadly no luck, the dropdown menu still instaclosing when clicking on it, thanks anyway

jordicunillerarivera, 2024-08-17:
Okay, i make my research and appear to be a hyprland problem, it happens to other applications too, they put some fixes that we can try.

https://github.com/hyprwm/Hyprland/issues/1593

closed 2024-11-24 by github-actions[bot]