OrcaSlicer/OrcaSlicer #5174: Retraction Wipe Moves Not Generated

lhndo, 2024-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

LH Stinger

### How to reproduce

Load the 3mf and slice the file

### Actual results

Observe the wipe moves are missing.

[image]

### Expected results

Wipes are present.

### Project file & Debug log uploads

I have included the resulting sliced file along with the 3mf.

[wipe missing.zip](https://github.com/SoftFever/OrcaSlicer/files/15140223/wipe.missing.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

The issue is related to the printing profile. If I switch to a different one it works, but this one stops working. 
I have recreated it from scratch (starting from the klipper profile) three times. It works the first times then it stops, even if the profile is exactly the same as when it worked, and was not updated since.

The profiles are stored in a separate folder and Orca is launched with --datadir=C:\...
There are no duplicate profiles present elsewhere.

This issue has been reproduced by other users, and it has been very inconsistent and hard to figure out why it happens.

Arx1kage, 2024-04-27:
I'm still running the Orca 2.0 beta too, and that issue is still there. My installation is completely standard, and I'm using the default folders for storing my profiles. Just like LH, there's no duplicate profiles anywhere, and when I switch to another stock Orca profile the wipes work fine for one slice, and then they stop working on that profile too. 

[RPI 90deg Mount_PLA_1h27m.zip](https://github.com/SoftFever/OrcaSlicer/files/15140243/RPI.90deg.Mount_PLA_1h27m.zip)

lhndo, 2024-04-28:
[settingsconflict-wipe.zip](https://github.com/SoftFever/OrcaSlicer/files/15142363/settingsconflict-wipe.zip)
[image]

I wiped all the settings, moved everything to the default install and the issue kept coming back.

In the end I traced the issue to n
when the "orca_presets" get created and that .3mf file is present in recent, the wipe retract breaks

Arx1kage, 2024-04-28:
Wait sorry. Would you be able to explain this further, or confirm if my understanding is correct? 

Basically, you're saying that if you've sliced something with the exact same settings as something you've done before, that's what breaks retractions?

lhndo, 2024-04-28:
> Wait sorry. Would you be able to explain this further, or confirm if my understanding is correct?
> 
> Basically, you're saying that if you've sliced something with the exact same settings as something you've done before, that's what breaks retractions?

No idea what's going on but there's some conflict of settings that stems of how things are loaded from the config file. Try baking up your config and then delete it. Start Orca and go though the initial wizard and see if wipes start working again. At some point things will break again. Wipe is one of them.. I don't know if there are other profile settings that don't apply, but would be great of someone takes a look of what's happening.

lhndo, 2024-05-03:
[image]
I believe the issue was with some profiles which had these overrides enabled, and which were pretty hard to spot.
I'll close the issue and monitor the situation.

closed 2024-05-03 by lhndo