OrcaSlicer/OrcaSlicer #5176: G-code error causing printing to stop

enjoyneering, 2024-04-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.0.0 Official

### Operating System (OS)

Windows

### OS Version

Windows 10 x64

### Additional system information

CPU: Xeon E5 2660 v2
RAM: 64GB
GPU: Radeon

### Printer

FLASHFORGE Adventurer 5M

### How to reproduce

After cutting out the part and printing over 80% it stopped. The printer log shows an error:

**Unable to parse move 'G1 X10.598 Y-45.945 00754'**

After downloading the g-cod from the printer (it was sent to the printer via wifi), the same line was found:

;WIDTH:0.290573
G3 X10.508 Y-45.881 I.94 J.642 E.00215
;WIDTH:0.330098
G1 X10.598 Y-45.945 00754    <--- HERE
G1 F6988.174
G1 X13.02 Y-45.809 E.0137
G1 F7078.551
G1 X13.248 Y-45.669 E.00822
G1 F7200

### Actual results

G1 X10.598 Y-45.945 00754 

### Expected results

Looks like it should be **E.0754** instead of **00754**

### Project file & Debug log uploads

The log file and project were not saved. There is only g-code.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-05-09:
Please attach your project file

enjoyneering, 2024-05-24:
Hi igiannakas,

I have only corrupted g-code. See line 274269.
[PlanetCarrier24_PLA_1h39m.zip](https://github.com/SoftFever/OrcaSlicer/files/15432319/PlanetCarrier24_PLA_1h39m.zip)

igiannakas, 2024-05-24:
Please attach the 3MF file…

enjoyneering, 2024-05-24:
Will do today. Thank your help.

enjoyneering, 2024-05-24:
Please see 3MF file in the attached
[PlanetGear24.zip](https://github.com/SoftFever/OrcaSlicer/files/15440767/PlanetGear24.zip)
 zip file

igiannakas, 2024-05-28:
Nope - attached is my sliced GCODE file straight from your project file
[PlanetCarrier24_PLA_1h10m.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/15473369/PlanetCarrier24_PLA_1h10m.gcode.zip)

Could you also take a look? Maybe try sending it for a print?

I'm using the 2.1-alpha version so the issue may have already been fixed. Maybe download and try the nightly release.

monkeyWad, 2024-06-11:
I got this error with a print a couple days ago. Here's my error code and gcode. Unfortunately I did not save the 3mf file that I used to generate the Gcode. I'm running version 2.1.0-beta

[Stacking game ninja mice v6_PLA_2h59m.txt](https://github.com/user-attachments/files/15788857/Stacking.game.ninja.mice.v6_PLA_2h59m.txt)

This is the full error about 75% into the print;
Unable to parse move 'G1 X-9.85-46.878 Y-77.4 I4.394 J-3.386 E.04835'

enjoyneering, 2024-06-14:
Flashforge has started pushing out new a5m firmware via OTA. A quick analysis shows that the Wi-Fi driver has been updated. Hopefully this will fix the issue. Will continue posting.

monkeyWad, 2024-06-21:
Just received this error again Unable to parse move 'G1 X51.59.746 E.00814'
I have the 3mf and gcode this time.
[Error.zip](https://github.com/user-attachments/files/15922011/Error.zip)

enjoyneering, 2024-06-23:
Hi monkeyWad,

What version of Orca are you using?

monkeyWad, 2024-06-23:
> Hi monkeyWad,
> 
> What version of Orca are you using?

The above was sliced on 2.1.0-beta. I've updated to the 2.1.0 release, but I haven't done much printing since.

closed 2024-09-29 by github-actions[bot]