OrcaSlicer/OrcaSlicer #5179: "Bottom surface flow ratio" only changes surface, not walls.

stueble, 2024-04-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu

### Additional system information

Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz

### Printer

Voxelab Aquila (Ender3 clone)

### How to reproduce

1. Change "Bottom surface flow ratio" to, e.g., 1.5
2. Slice Model
3. Have a look at the flow of the first layer. 

### Actual results

The flow of the surface between the walls have been changed, but the flow of the first layer walls stay the same.

### Expected results

Since this option is under "Walls and surfaces", and since there is no other option to change the first layer flow of walls (and brim, and skirt, etc), this option should change the flow of all first layer lines.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

If this option only changes the infill by intention, there is definitely an option missing to change the flow of the first layer walls, skits, brim, etc, since this is most often the problem, not the infill.

closed 2024-08-05 by github-actions[bot]

cron410, 2024-12-26:
I have this same issue. All other flows are spot on, the brim seems to be very low flow.

jasc76, 2025-08-20:
I have the same issue. Brim Lines dont touch and the brim separates from the print and doesn't provide the support from lifting as needed
All other first layer stuff is fine, but Brim is not usable