OrcaSlicer/OrcaSlicer #5185: Extra walls being generated by modifier

foresterr, 2024-04-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality Ender 3 v3 KE

### How to reproduce

Create a very simple model - for example a cube with a vertical hole through it (can be done entirely by use of primitives without leaving Orca). Add a modifier which has the entire hole inside it, but does not touch outer cube walls. Change any modifier setting related to walls (for example number of Wall Loops). Slice the model.

### Actual results

Additional walls are added at the modifier boundary.

### Expected results

Walls inside the modifier are altered according to the modifier (e.g. they have a different number of wall loops), but no extra walls are generated in the sliced model where the modifier ends. Maybe this is working as intended to avoid broken wall loops if the modifier boundary intersects a wall loop, but in case of wall loops fully enclosed inside a modifier, the result is rather confusing.

### Project file & Debug log uploads

[modifier potential bug.zip](https://github.com/SoftFever/OrcaSlicer/files/15142574/modifier.potential.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

There is another bug issue (https://github.com/SoftFever/OrcaSlicer/issues/3097) that might be about the same thing, but it's closed as stale and tbh I can't figure out if the reporter there talks about the same thing (it's a bit confusingly written and illustrated).

FDFD1313, 2024-07-08:
Same here, the modifier adds walls, instead of only changing the region walls, (is changing the region and, additionally, introducing new undesired walls)
[image]

ProfessorKaos, 2024-08-09:
PIng to this

TylonHH, 2024-08-18:
yeah that realy sucks
[image]

Misterke, 2024-08-29:
Bump.  Note that this problem also exists in upstream PrusaSlicer.  That doesn't make it any less annoying though.  It's almost impossible now to use fuzzy skin but have horizonal screw holes excluded from the fuzziness.  Having a modifier over those holes adds annoying walls with the most outer ones still having fuzziness applied.

Peteeeeeeeeee, 2024-09-17:
also exists in Bambu Studio and is driving me insane.

irowboat, 2024-09-18:
I think this is what I’m seeing with my multi-color print right now. As a simple test, I’ve painted some spots on the outside surface of the object, and the colors run into the layer as I’d expect. What I **don’t** expect is for each of the _**internal boundaries**_ between colors to have extra perimeters or fuzzy skin. It seems like color is just another modifier as far as Orca is concerned, and is generating these internal walls—_with all the parameters_—drastically slowing the print (doing fuzzy skin on internal perimeters especially).

This has 4 perimeters (more perimeters = more fun) and a really fuzzy skin (0.02mm/1mm). It isn't dependent on classic or Arachne from what I can tell, but WOW does it affect the new beam interlocking if you've also got fuzzy skin:

[image]

without beam interlocking

[image]

**_with_** beam interlocking

#### Example file
[OrcaSlicer - issue #5185.3mf.zip](https://github.com/user-attachments/files/17047150/OrcaSlicer.-.issue.5185.3mf.zip)

Ishprecher, 2024-11-23:
This is a problem that dates back to Slic3r apparently. Probably won't be fixed on Orca or Bambu until Prusaslicer does, but isn't likely since they've known about it for years now.
https://github.com/prusa3d/PrusaSlicer/issues/8121

closed 2025-03-01 by github-actions[bot]

BSKFeralLycan, 2025-04-30:
Yeah it's still happening it's happening on Orca ports too It also happens on purpose laser I don't know why they haven't fixed this yet.. I'm also confused on why they've already closed your issue here?

Have you seen my photo there set for screw holes in three they're unmodified and the one that I'm trying to reinforce has created its own section this is also extremely annoying when you're just trying to do something like fuzzy skin because even though you're just trying to deal with the outside walls it's still creates a perimeter and mind you I have zero perimeters chosen so this should be zero. 

Off topic but I really wish that they would just make a paint tool for fuzzy skin and modifiers that only affect the outer perimeters

[image]

Misterke, 2025-04-30:
Latest Orca at least fixed the problem when having a modifier just applying fuzzy skin and also when just modifying the amount or type of infill, but if you change the number of walls (and probably some other parameters affecting walls) then indeed additional walls are added at the boundary of the modifier.  I myself am already very happy that for infill and  fuzzy skin I can now just create modifier objects in CAD without then getting extra walls weakening those parts ...

TylonHH, 2026-10-24:
> 19\. What I also just noticed: if you only have the modifier change things that don't add walls, slicing gives a nice result without extra walls, BUT ... if you add a parameter to the modifier which does add those extra wall, slice and then remove that parameter again (so restore it to default), then still extra wall are generated when you slice - even though the settings are the same as what before was not giving you extra walls.  If you then disable the entire Assembly and then reenable it and reslice, then you get back to not having those extra walls ...

I just wanna just slow down the outer wall in a specific region, this also leads to a extra internal walls from modifier 😭