OrcaSlicer/OrcaSlicer #5188: Allow editing of acceleration for travel when klipper g-code flavor is selected

Gozania, 2024-04-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is asinine that travel acceleration cannot be edited.  I understand this for marlin and such as a saftey precaution, however those of us who are on klipper should have total control of what OUR machine does.  My ender 3 V2 can SAFELY do 5Kmms travel acceleration, yet the field is greyed out and capped at 500mms.  I do not want to travel this slow...
[image]

### Which printers will be beneficial to this feature?

Klipper, All

### Describe the solution you'd like

If klipper is selected open up ALL motion fields for editing.  Perhaps make a check box that needs to be ticked stating that the user understands the risk of editing the fields from default.

### Describe alternatives you've considered

1. Go back to PrusaSlicer until this is remedied along with other bugs I have reported.  

### Additional context

I really want to use your software due to its reputation and layout.  However there are things I cannot overlook that orca slicer does that prusa does not.  A 3rd party slicing software should NEVER limit people on what they can and cant do with THEIR machine motion ability.  C'mon Softfever, you can do better than this.

jamincollins, 2024-04-28:
1. Tone it down, jeez. There is no need (or call) for the tone in the above report.
2. If you hover over the setting you'll find that it is disabled because it only applies to Marlin.

[image]

jamincollins, 2024-04-28:
Here's the section of code that disables the field if you're using Klipper or legacy Marlin:
https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/Tab.cpp#L4392-L4399

Gozania, 2024-04-28:
I don't know what to do with the below link.  I dont know how to code or program.  Apologies for my frustration initially, but why is this a default when klipper is selected?  Marlin, I can totally understand & would encourage the current configuration.  While helping delvelop an S1 Plus profile for orca the developer I was helping also ran into an issue where he was unable to increase travel acceleration for that machine.

What needs to be done on my end to achieve my desired results?

https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/Tab.cpp#L4392-L4399

Gozania, 2024-04-28:
Never mind the instructions... I figured it out.  Was able to achieve it by switching to marlin 2, making my adjustment, then switching back to klipper prior to saving.

https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/Tab.cpp#L4392-L4399

jamincollins, 2024-04-28:
Your change has no meaning.  The setting is only used on a Marlin 2 system.  This is not a bug.

Gozania, 2024-04-29:
Ok, then how can I make it so that change does have meaning and will reflect while printing?

mxfi, 2024-04-29:
Klipper doesn't use max travel acceleration limits like in marlin. You can google how printer limits are set up and why, the information is readily available anywhere really.

What you have there is machine limits. Klipper sets this machine limit in the config file and doesn't accept changes from slicer, it has nothing to do with orcaslicer. It doesn't matter because as the popup menu explains, it's only used in Marlin 2 and is null in Klipper. 

You can change travel acceleration in the printing profile "speed" tab. You can also set a max limit to any movement in your Klipper printer.cfg file. Doesn't matter if orca "maximum acceleration for printer" is set to 1 or set to 10000, if you put travel acceleration as 5k, it will move at 5k accelerations as long as it doesn't exceed your Klipper config limit, which is why it's greyed out.

closed 2024-08-05 by github-actions[bot]

unitrix0, 2024-11-11:
Two things I want to mention:

1. In version 2.2.0 there's no more pupop message on hover. At least not on windows.
[image]
2. Why even showing this field if it's only Marlin related and the printers flavor ist set to Klipper?

Benik3, 2025-01-07:
I can confirm that the tooltip is not showed anymore and that's why I ended here :)

VORONENTHUSIAST, 2026-02-19:
Over a year late on this one, but only just started using Orca Slicer - Change firmware flavour to Marlin 2, Max Accel for Travel becomes editable - Set it to what you want - Change FW flavour back to Klipper, box is greyed out but the stupid warning goes away.

According to Mainsail, Klipper is now adjusting to the higher accel I've set in the speed section of Orca Slicer when travelling.

I'm sure one of you must've figured this out by everyone saying it's linked to the Marlin 2 FW, but if you didn't - you're welcome!!