OrcaSlicer/OrcaSlicer #5190: Printer search doesn't look for Brand names

Scootworks, 2024-04-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-dev

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4.1

### Printer

none

### How to reproduce

1. Hit Printer gearbox icon
2. Enter device keyword: Prusa
3. nothing appears -> use the same

### Actual results

My guess is, that the search bar is only looking for printers and not categories
[image]

### Expected results

Maybe we should add either the brand name to the printers or it would be nice to list the brand names including all devices

### Project file & Debug log uploads

not required

### Checklist of files to include

- [X] Log file
- [X] Project file

tcurdt, 2024-07-28:
Stale or not - still a bug.

tcurdt, 2024-10-27:
still a bug

closed 2025-02-03 by github-actions[bot]