OrcaSlicer/OrcaSlicer #5194: build fails on MacOS 14.3 arm64 The link interface of target "opencv_world" contains:      Eigen3::Eigen    but the target was not found.

superfloh247, 2024-04-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

226450e

### Operating System (OS)

macOS

### OS Version

14.3.1

### Additional system information

arm64
latest XCode

### Printer

Bambu P1P

### How to reproduce

git clone https://github.com/SoftFever/OrcaSlicer.git
cd OrcaSlicer
./build_release_macos.sh

### Actual results

CMake Error at deps/build_arm64/OrcaSlicer_dep_arm64/usr/local/lib/cmake/opencv4/OpenCVModules.cmake:81 (set_target_properties):
  The link interface of target "opencv_world" contains:

    Eigen3::Eigen

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  deps/build_arm64/OrcaSlicer_dep_arm64/usr/local/lib/cmake/opencv4/OpenCVConfig.cmake:126 (include)
  src/libslic3r/CMakeLists.txt:479 (find_package)

### Expected results

build succeeds 

### Project file & Debug log uploads

not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lbibass, 2024-04-30:
I am also seeing the same problem. I've installed openCV locally, as well as eigen.

superfloh247, 2024-04-30:
Local installation of openCV and Eigen: root cause of the problem or solution (workaround)?

superfloh247, 2024-04-30:
I can answer my question: removing locally installed (by brew in /opt) eigen fixed it, the build works again for me

ldr, 2024-04-30:
Ran into same issue here as well while trying to install bambustudio-git from AUR on arch

ldr, 2024-04-30:
> I can answer my question: removing locally installed (by brew in /opt) eigen fixed it, the build works again for me

updates happening in realtime :) tnx will read up if local install exists

lbibass, 2024-04-30:
uninstalling eigen locally didn't fix my problem.

ldr, 2024-04-30:
Didn't fix the build myself here, had eigen installed locally for another app and couldn't remove it. Worked around it by installing bambustudio from flathub, which worked immediately :) https://flathub.org/apps/com.bambulab.BambuStudio

PR #5311 (2024-05-10): disable eigen for OpenCV
Fixes #5194