OrcaSlicer/OrcaSlicer #5199: Flow calibration scaling up with 1mm nozzle

Daitallica, 2024-04-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Creality Ender 5

### How to reproduce

Set nozzle width to 1mm nozzle,
click calibration
click flow rate pass 1
Click preview
OR
Click Slice

### Actual results

Models scale up massively in X, Y and Z and no longer fit on the plate

[image]

### Expected results

Models to stay at same scale

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15152714/log.zip)
[Pre and post slice project.zip](https://github.com/SoftFever/OrcaSlicer/files/15152715/Pre.and.post.slice.project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Ive seen this issue raised and closed at least twice on github with no fixes, I found these git reports on the orca slicer discord.

https://github.com/SoftFever/OrcaSlicer/issues/449
https://github.com/SoftFever/OrcaSlicer/issues/3729

Daitallica, 2024-05-14:
I think have figured out a workaround

Select all (ctrl + a)
Right click => convert from inches
Right click => restore to inch

it appears to bring them back to on load sizes

mleone87, 2024-06-17:
same for 0.8
still present on 2.1.0-rc

Daitallica, 2024-06-17:
> same for 0.8 still present on 2.1.0-rc

Out of curiosity did you try the workaround I posted?

mleone87, 2024-06-17:
> > same for 0.8 still present on 2.1.0-rc
> 
> 
> 
> Out of curiosity did you try the workaround I posted?

Yes and the scaling disappears but something odd remains in the slicing, it's messed somehow

closed 2024-09-24 by github-actions[bot]

sleewok, 2024-10-30:
This is still a problem with the latest 2.2 release. @Daitallica thanks for your workaround!

DaKrampus, 2025-01-09:
Strange: i also ran into the bug with 2.2.. but the strange thing is: it works as it should with the 0.6 nozzle. The bug appears with the 0.8 nozzle. The workaround of @Daitallica seems to work. You are greatI also tried re arranging the objects .. that also seems to work.. is it really precise? i dont know. but someone should post this as a persisting bug.. here it is useless as the issue is closed!