OrcaSlicer/OrcaSlicer #5201: Small area flow compensation

jonny7737, 2024-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

14.4.1

### Additional system information

CPU: M2 Max
Memory: 64GB
Video: M2 Max

### Printer

Creality K1

### How to reproduce

Add part to plate
In Quality, enable Small area flow compensation
Slice plate
BOOM!!!Crash

### Actual results

Crash

### Expected results

Slice

### Project file & Debug log uploads

[3D Benchy.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/15156264/3D.Benchy.3mf.zip)
[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/15156265/logs.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

faith-healer, 2024-05-03:
exactly the same here

OrcaSlicer 2.0.0+ (latest 2.1.0 nightly as well)
TT Sapphire SP-5
AMD Ryzen 5800X / 32GB RAM / RTX3080ti
Win10 / Arch KDE6 X11

Spekkie, 2024-06-07:
I have the same problem...
Disabling "arc fitting" in settings is a working workaround for me so far.

RimoVR, 2024-06-09:
Can confirm what Spekkie said. Arc fitting doesn‘t currently seem to be compatible with small area flow compensation. Maybe the flow compensation is running after arc fitting right now?

closed 2024-09-15 by github-actions[bot]