OrcaSlicer/OrcaSlicer #5204: 3Dconnecxion devices support on Linux

buzzhuzz, 2024-04-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

OrcaSlicer has support for 3Dconnexion devices on Mac and Window, but not on Linux.

This feature was requested by @discip 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

3DConnextion device could be used with OrcaSlicer on Linux.

buzzhuzz, 2024-04-30:
I got my hands on 3DConnection SpaceNavigator Wireless and found that all the code needed already there.

The only thing is need to be done is libspnav-dev added to build dependencies in order to have 3Dconnection code enabled.

Steps I've done in order to get it working:
# Add ```libspnav-dev``` to the bulid dependecies and rebuilt the OrcaSlicer
# Install ```spacenavd``` on the host system

Things need to be considered while enabling this feature:
# ```spacenavd``` is really optional. Applications built with ```libspnav-dev`` would run fine with no daemon present in the system.
# ```libspnav`` either shall be bundled into AppImage/flatpak or linked statically.
Current cmake scripts link ```libspnav``` dynamically

discip, 2024-04-30:
# @buzzhuzz
# Great! Finally! Thank you so much! 😃🤝🎉🚀❤️
Will test as soon as I'm back on my Linux machine.

buzzhuzz, 2024-04-30:
Test build available here https://github.com/buzzhuzz/OrcaSlicer/actions/runs/8892413761

I still need either https://github.com/SoftFever/OrcaSlicer/pull/5203 or https://github.com/SoftFever/OrcaSlicer/pull/4175 integrated before making PR per this task.

discip, 2024-05-08:
@buzzhuzz
Unfortunately I'm not able to run the AppImage you provided.

ARDev1161, 2024-06-21:
@buzzhuzz What about PR? =) Did you make it?

discip, 2024-09-20:
@buzzhuzz
Any news so far?

AMstuff, 2024-10-26:
@buzzhuzz Are you still on to this? I have just built a flatpak with libspnav support ([commit](https://github.com/SoftFever/OrcaSlicer/commit/e89cde9f9d1ef986df81e6cd8fef1c60cd83c71f)) and it works flawlessly for me.

PR #7529 (2024-11-28): Include libspnav in flatpak builds
# Description
This PR includes libspnav in the flatpak builds. Allowing usage of 3Dconnexion devices on linux systems with spacenavd is installed.

This partially fixes #5204 . However I'm only including my patch for flatpak, as I didn't confirm that the additional patches by @buzzhuzz ([commit](https://github.com/buzzhuzz/OrcaSlicer/commit/141b7c8154abe1bebac99478598c07ba9b45afd0)) are working.

## Tests
Built and verified working on my system.
closed 2024-11-28 by SoftFever