OrcaSlicer/OrcaSlicer #5206: user profiles  gone every 2nd start

Acuriz, 2024-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10 22H2

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

After updating to 2.0.0 i mentioned that my profiles for my printer including filament profiles were gone -> They were imported before.

i did a system restore to get them back.

i did this several times  but it was always the same behaviour. then i mentioned that every second time i start orca they are shown. 

### Actual results

read "how to reproduce"

### Expected results

profiles are always there.

### Project file & Debug log uploads

[Error_log_and_projects.zip](https://github.com/SoftFever/OrcaSlicer/files/15158840/Error_log_and_projects.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Dsema, 2024-05-13:
I’m experiencing the same issue. I’m working with Fusion 360 to design some parts and when I try to export them directly to OrcaSlicer, half the time the user profile I design does not load. 
When I notice the issue, I attempt to open Orcaslicer multiple times using the shortcut I typically use and discover that it has the same problem.

Acuriz, 2024-05-13:
its quite annoying. but at least there is a workaround by always starting and closing orcaslicer between the sessions i need. its weird how it is possible that it is exactly every 2nd start.

closed 2024-08-20 by github-actions[bot]

ElektroCoder, 2024-09-13:
I’m experiencing the same issue.