OrcaSlicer/OrcaSlicer #5217: "Avoid crossing walls" feature not working.

MAJCcustoms, 2024-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1

### How to reproduce

Turn on avoid crossing walls
set the max detour length insanely high

### Actual results

Avoid crossing walls still does not change travel moves and crosses walls anyway.

### Expected results

Avoid crossing walls avoids crossing walls

### Project file & Debug log uploads

Unfortunately cannot include a log as I am working on proprietary models.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I believe I have found at least one source to this bug. I can fix this issue of avoid crossing walls not working by disabling "only one wall on top surfaces", but I'm not sure what the actual cause of this setting breaking avoid crossing walls is.

sizzname, 2024-05-06:
Confirming the problem. It does not depend on the printer model. Max detour length does not affect the result either.
The screenshot shows the resulting travels.
[image]

gitvisual, 2024-05-25:
+1 on this. Just tested a bit and for me it has trouble with avoiding walls going INTO the infill. it avoids walls when the infill is done though. See image:

[image]

Only one layer shown. This is from a simple hollow pipe model.

gitvisual, 2024-05-25:
FOUND THE CULPRIT!

[image]

If this setting is turned off, the "avoid crossing walls" works as intended. Maybe by design for some reason?

LitlSkitl, 2024-06-01:
Can confirm, having the same issue here. Just noticed as I'm printing here.
[image]

And goddammit the left clip got knocked off. Wow.

alfondehesa, 2024-06-29:
+1 Here. Exactly the same problem, once it's done with the walls and goes to start the infill, it ignores walls. 
[image]

antidotcb, 2024-09-16:
Not working in 2.1.1
Crossing top surfaces and making scratch on them as result
[image]

convicte, 2024-10-28:
Apart from obviously the maintainers time, which is a precious resource, what else is required for this to be picked up for review?

After having realized how much of the stringing and artifacts this is causing, I was hoping to assist in sorting this.

Thanks in advance!!

SlipstreamFPV, 2024-12-15:
Same issue. I hope this will get looked at. It really ruins TPU parts. Anyone got a workaround?

[image]

wociscz, 2024-12-15:
Ha :) My actual TPU print, same problem.
I'd call this option "Move within walls" to move only inside walls and with an option for top surfaces too (to avoid scratches on top surfaces).

real-artswan, 2025-01-11:
Confirmed on 2.2.0. "Max detour length", "Only one wall..." options don't affect. Can be easily reproduced with an embedded "Flow rate" calibration prints.

Kellwaxx11, 2025-01-17:
It seems that the issue is present only when using "Arachne" wall generator , if you use "Classic" the function works fine 

[image]

[image]

rmeissn, 2025-02-05:
In my opinion, "Avoid crossing walls" is not enough at all. In the below screenshot, the last travel move is going directly through already printed supports, which might knock them off. This happens regardless of the Classic or Arachne engine and causes problems with tree supports, which are easier to knock off or break. Other travel moves actively avoid already printed parts (including supports), while other moves go through already printed parts (including supports) while this would be avoidable, which is kind of inconsistent.

[image]

c-ried, 2025-02-18:
Hi everyone! I'm on version 2.2.0 and observe the same issues described above. But none of the workarounds stated here works as you can see in the screenshot below. Neither Classic nor Arachne wall generator combined with the "only one wall.." options deselected results in the desired behavior. 

I just migrated from Cura to Orca Slicer for various reasons. But in Cura this feature works and can be fine tuned as "combing". Would be great to see Orca Slicer to offer this feature, too!

[image]

c-ried, 2025-02-18:
Just noted that the feature works but only in combination with a retraction! By lowering my travel distance threshold (see below), retracts and unretracts are introduced and the travel paths are within the outer walls region.

In my use case, I intentionally want to reduce retracts in combination with scarf joint seams. So using a longer travel distances is a must for me. Any chance to have "avoid crossing walls" show the same behavior even when not retracting the filament?

[image]

real-artswan, 2025-02-24:
Just to be fair, if nozzle moving over print knocks off something then it means that your prints are overextruded, it's more thick than it should and that printer "think" it should. Try to dry plastic and calibrate flow rate coefficient.
Regardless of the issue though...

DanyloBilyk, 2025-02-24:
> Just to be fair, if nozzle moving over print knocks off something then it means that your prints are overextruded, it's more thick than it should and that printer "think" it should. Try to dry plastic and calibrate flow rate coefficient. Regardless of the issue though...

It's not always over-extrusion. Consider these additional factors:

Steep Overhangs:

In some cases, steep overhangs cause the ends to curl up slightly due to imperfect cooling, overheating, or surface tension.
Air Bubbles:

A blob can form if an air bubble gets trapped in the plastic.
Filament Moisture:

Using filament that isn’t completely dry may result in irregular extrusion.
Retraction Issues:

Slight inaccuracies in retraction—especially with a slick Diamant nozzle—can lead to a small drip.
Infill and Wall Overlap:

An incorrect overlap between infill and walls may produce an uneven top surface with bumps.
Filament Diameter Variations:

Variations in filament diameter can cause unexpected extra extrusion.
When the nozzle passes over these bumpy areas, it can hit and dislodge them. If supports are robust, the print might be fine; however, weak supports (such as thin, underextruded branches with poor bed adhesion) can be easily knocked off or broken, depending on the weight of the toolhead and the speed/acceleration. This is why a z-hop is often recommended—even though sometimes its height or trajectory isn’t set optimally.

Overall, while over-extrusion is a common culprit, other factors can also lead to these issues.

MxBrnr, 2025-03-08:
Confirmed still an issue in v2.3.0-beta2.
/home/mxbrnr/Desktop/Flowrate Test - Pass1.3mf.zip

[image]

real-artswan, 2025-03-10:
Overall, from experience: travel is very underoptimized in the Orca

[image]

[image]

it makes stupid travel moves all the time, sometimes travel takes up to 20% of print

carlosjln, 2025-04-16:
Is it me, or there's a correlation between "travel distance threshold" not triggering retraction (as expected) and "avoid crossing walls" not considering it is still a travel move and it has to go through the inside of the model?

[image]

RF47, 2025-07-20:
try this https://github.com/SoftFever/OrcaSlicer/pull/10185

closed 2025-07-27 by Noisyfox

yazz007, 2025-10-02:
I'm on 2.3.2 dev nightly and I don't think this patch did anything to enable proper movement. Or maybe I need some special settings? I tried each setting mentioned but I can't force to avoid crossing walls.

[image]

RF47, 2025-10-02:
> I'm on 2.3.2 dev nightly and I don't think this patch did anything to enable proper movement. Or maybe I need some special settings? I tried each setting mentioned but I can't force to avoid crossing walls.
> 

Make sure you don't have a timelapse-gcode,  it is a necessary condition for it to work avoid crossing perimeters.

yazz007, 2025-10-02:
I never applied timelipse, not even know how to do it ;-)

RF47, 2025-10-02:
I think this is a new bug related to the part size. If I scale it to 115%, it works fine.
We'll have to investigate further to find out what the problem is.

[image]

[image]

yazz007, 2025-10-02:
Thanks! I can confirm the same, 115% :) is the sweet spot :)

RF47, 2025-10-02:
only X axis scaling is needed to work properly.
[image]

PR #10185 (2025-07-27): Bug fix: avoid crossing perimeters
This pull request partially addresses the issue described in #5217.

Problem:
At the end of a perimeter, after retraction, the following travel move does not respect the restriction that prevents crossing other perimeters. 

Notes:

Works best when using Arachne perimeter generator.

Spiral Z-hop should be avoided for better results.

## Tests
[image]
[image]
[image]
[image]
[image]

this PR may close #5217 
this PR may close #9770