OrcaSlicer/OrcaSlicer #5224: Preview Pane - Huge performance issue

Alkerion, 2024-05-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Huge performance issue with Preview pane.
If Preview Pane is displayed in the slicer the GPU usage rise up to 70%, even if we don't move anything.
Worse, the GPU usage stay at 70% even if we minimize the window.
PC become sluggish, take ages to go back to Prepare Pane or to change a setting.

Less of a problem with ultra high end GPU (E.g. RTX 4090), but a bit overkill and over expensive for a slicer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Less GPU usage, better perfs, better render engine.
No high GPU usage when nothing move.
No high GPU usage when window minimized.

DVSVIDEO, 2024-05-01:
I join. I have also noticed this problem.
On my 2 monitor configuration, the video running in window 2 starts to snippet when the preview window is active.
Something is eating the performance even when nothing is happening....

tlhintoq, 2024-05-02:
I found this makes a huge difference.  Set Windows to assign the high-performance GPU to OrcaSlicer in the advanced graphic display options.  
[image]

I kinda wonder if the recent round of Windows updates has left it not switching from the Intel chipset to the high-performance card on demand.

Sidebar:  Motion control with the mouse is smooth but with my 3DConnexion Space Navigator 3d mouse Orca stutters.  Not life-ending or anything, but definately noticable stutter when moving, zooming, spinning etc.

tlhintoq, 2024-05-02:
> That's more or less what I face, you need a >800$ GFX card to have something more or less fluid.

I'm on a laptop with a 3080mobile.
[image]

But I'll back up and ask... 

1. What _do_ you have for a video card?
2. Have you selected Orca to use that card within Windows as in the earlier screenshot, so it is forced to use the best you have whatever that might be?

TomPcz, 2024-05-02:
Perhaps duplicate of #3899?

Alkerion, 2024-05-02:
When there is only one GFX on the PC there is no need to define anything, not useful too on a desktop PC, where the monitor is connected to the GFX directlly.

I've several PC, the one I use the most (located in the printers room) as an integrated Intel UHD.

My work PCs have a GTX 780Ti and a RTX 3080Ti, and my gaming PC a RTX 4090.

I've tried with a "very simple" model, 4000 triangles :

- Intel UHD 100%
- 780Ti 100%
- 3080Ti around 90%-95%

I'm not sure it's normal that it's required to have such high end GPU to have something fluid.
[image]
[image]

Alkerion, 2024-05-02:
> Perhaps duplicate of #3899?

Not at all, my concern is about the Preview Pane only, not global slow down, and in my case the GFX usage is stratospheric.

DVSVIDEO, 2024-05-03:
I have an older processor, an i5 4440S 2.08Ghz
with an integrated video unit: HD4600

This is a very old (10 years old) processor, but 
it is perfectly enough to drive 2 monitors, 2 monitor video editing with GrassValley Edius broadcast editor (499 EURO)
Enough to run Autodesk Fusion 360 smoothly
Enough to play fullscreen video on 2 monitors at the same time 
But not enough for a gcode display?
There must be some optimization error.

fooshards, 2024-05-20:
Adding to this:

Steps to recreate:
Slice a model.  Large ones are better for showing the issue.
Be sent to the preview pane [defaulting to "Line Type" rendering].
Select the "Speed" render preview.

this setting will cause your GPU and CPU to pin at very high usage even when idle.  

Without looking at the code, I'm guessing it is re-calculating endlessly, not just when the camera preview moves.

evilhamsterman, 2024-05-27:
I'm seeing similar issues, I just sent a very basic ring to the preview and while it doesn't peg it does consume a significant amount of CPU and GPU

J3LS, 2024-06-01:
Just in case this is of help to the devs. 

I noted above other posters appear to have better or specific graphics cards and wondered whether this was connected to the issue as I have a very basic setup and although there is some effect the machine continues to work fine.

I do notice high cpu and memory usage but only after an stl is added, however apart from the cpu fan going bananas the machine remains usable with no noticeable slowdown elsewhere. On first run, if left idle, cpu usage is so low as to be off the scale and memory usage is 1Mb.

System:
  Kernel: 5.15.0-107-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Xfce 4.18.1
    tk: Gtk 3.24.33 info: xfce4-panel, plank wm: xfwm 4.18.0 vt: 7 dm: LightDM 1.30.0
    Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: Dell product: Latitude E7440, 16Gb memory
  Info: dual core model: Intel Core i7-4600U bits: 64 type: MT MCP smt: enabled arch: Haswell
    rev: 1 cache: L1: 128 KiB L2: 512 KiB L3: 4 MiB
Graphics:
  Device-1: Intel Haswell-ULT Integrated Graphics vendor:

tiivonen, 2024-06-08:
This is an issue with MacOs also. Happens with basic M1 and M3 Max. When preview window is open GPU usage goes sky high. This happened with previous releases too.

Alkerion, 2024-06-19:
How can I turn that as a BUG ?

J3LS, 2024-07-28:
Possible connection to this issue.

Background
I've just swapped ISP so changed router. Even though Orca has the new credentials I can't connect so back to a flash drive again. I can live with that until I solve it but...

Until now (with a good connection to my Ender 3 V3 KE) just leaving Orca running but not being used for a few minutes resulted in high cpu usage and the fan going bananas. Now I have the "Could not connect: No route to host" on the device page but leaving Orca running for any amount of time does nothing. CPU etc. remain idle and all quiet.

Sorry, I can't verify this by connecting again! (For the moment).

PhapandNap, 2024-08-27:
I have my office equipment on solar, so I am monitoring my power usage a lot more. I noticed this today. I had 3 projects open for my 3 printers and when I was on the preview view my PC was using close to 500w. I heard the fans kick up and stay on for a bit. Even with the windows minimized. 
I just start prints and close sessions now, but I don't remember this happening in past versions, but I could have just not noticed. The fans are a dead giveaway though and I think I would have noticed. 
The 3D graph section on the GPU is at 95% when this happens as well.

closed 2024-12-03 by github-actions[bot]