OrcaSlicer/OrcaSlicer #5225: Modell slicing by height for different layer height, like Simplify3D

DVSVIDEO, 2024-05-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is currently not possible to specify different layer heights within a model.
Missing layer height, seam position, when splitting a body with a modifier or a plane cut.
This is the only thing missing since I switched from Simplify3D...

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When a model is cut parallel to the bed (horizontally) using the flat cut option (preparation top menu bar)
Then there is nothing to prevent you from setting a different layer height or even a different seam placement above and below the cut.
Quite understandably, if the cut crosses layers, this is not possible.

In Simplify 3D Slicer there is an option to divide into sections by specifying the heights and this is exactly what you can do.
Each layer can change everything.

[image]

### Describe alternatives you've considered

1. A modell betöltése
2. A cut (c) menü ikon kiválasztása
3. A planar opció kiválasztása
4. A sík megadása ablakban mm-ben, vagy a modellen megragadással és húzással
5. A részekre darabolás négyzet kipipálása
6. A Vágás gomb megynomása
7. A Folyamat felirat mellett a Tárgy kiválasztása
8. A modell neve_A és modell neve_B tárgy valamelyikének kiválasztása
9. A kiválasztott modell szakasz réteg magasságának a megváltoztatás (vagy a varrat pozíció)
10. Szeletelés

### Additional context

This is how Simplify 3D does it:

[image]

Arakon, 2024-05-01:
Do you mean the height range modifier that is available via right-click on the model?
[image]

DVSVIDEO, 2024-05-01:
Oops...
When did this get in?  I didn't notice the change.
As the only feature that is not translated into my native language, it was added sometime later...
thanks for your reply!
I'll close the request.

DVSVIDEO, 2024-05-01:
Already solved.

closed 2024-05-01 by DVSVIDEO