OrcaSlicer/OrcaSlicer #5229: Feature Request: Project Image onto Model.

JourneyTronix, 2024-05-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

easier way to get what you want using the paint feature.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

It would be great to be able to project an image onto a 3D asset for usability with the paint feature. Say I have an image of a Brick wall and I rasterize it so it only has 8 colors I could then project that onto the asset via the painting feature and then print with AMS loaded and rdy to go! 

### Describe alternatives you've considered

painting by hand.

kylek29, 2024-05-02:
There was some work done some time ago on PrusaSlicer to allow images to change the shape of objects (same way game engines do, e.g. normal/bump maps and tesselation) -- similar concept: https://forum.prusa3d.com/forum/prusaslicer/wip-displacement-map-feature/

JourneyTronix, 2024-05-02:
That's cool,  I was thinking this would just simply project an image like an overlay as a painting guide or even transfer it to the mesh via tri planar projection or world space etc.  Don't really want it to change the models geometry.

closed 2024-08-10 by github-actions[bot]