OrcaSlicer/OrcaSlicer #5233: Add Interface shells when using Multi Color / Multimaterial.

CyberManiaK, 2024-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The problem is that when using multimaterial the top shell layer instead of making a solid layer it leaves the space for the next object. So instead of having straight uninterrupted movements, it needs to stop the movement producing an uneven top layer. 
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The solution is to add the option of "interface shells" like Prusa/SuperSlicer have it. This option gives a nice continuous solid layer, ready to receive the next object.   
[image]
[image]

Alkerion, 2024-05-02:
I join this request, I was looking for this exact feature too.

igiannakas, 2024-05-02:
Is this it here?
[image]

ramorimdias, 2024-06-13:
This feature is already present on bambustudio when developper mode is activated. I can find many "interface shells" strings on the .po but cannot make it appear on Orca even with Developper Mode on. It's weird. It works flawlessly on bambustudio. This is a must if you print coaester, keychains or logos to have a smooth top layer before changing color.

closed 2024-09-20 by github-actions[bot]

soxhi8, 2024-10-07:
This feature would be very useful.

exen904, 2024-12-19:
Any updates on this?

rinkek, 2025-01-10:
Would like to see this as well.
Works very good in both PrusaSlicer and Bambu Studio (when developer mode is enabled)