OrcaSlicer/OrcaSlicer #5235: allow flow test customization

sedan-cell, 2024-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

no

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

similar to the other calibrations allow flow rate to be customized. sometimes pass 2 being only negative is not fitting or someone might want smaller steps if possible

pgifford, 2024-05-05:
> sometimes pass 2 being only negative is not fitting

The value from pass 1 should be such that a negative adjustment is required in pass 2.

I agree that customization would be nice, however.  For many filaments I could get by with 1 pass at +/- 3.  As it is I often run only pass 1; estimating an in-between value is good enough.

dewi-ny-je, 2024-06-06:
I think that for most filaments most printers won't need the first pass, if the second one is performed in 101%-91%.

For example I know that my printer always works in a certain range, so I would like to customise the calibration to have, for example, 10 steps in 0.5% increments, rather than 10 steps in 1% increments which I know I won't need.

As alternative, a third pass with 10 samples in 0.5% steps could be offered, even without the need to offer a customisation.

sedan-cell, 2024-06-07:
> The value from pass 1 should be such that a negative adjustment is required in pass 2.
>
> I agree that customization would be nice, however. For many filaments I could get by with 1 pass at +/- 3. As [...]

sure and time can be saved by not printing all the others on pass 1 

> I think that for most filaments most printers won't need the first pass, if the second one is performed in 101%-91%.
>
> For example I know that my printer always works in a certain range, so I would [...]

same. i know i usually am within 0 to -3 or -4 so i always have to delete a bunch of squares every time. if i can specify the range it saves time and filament

closed 2024-09-13 by github-actions[bot]