OrcaSlicer/OrcaSlicer #5243: Enabling "Independent support layer height" when printing with supports enabled and using different material for support interface causes support interface material to be used for the entire layer and all line types

enz1ey, 2024-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1, 2.0.0

### Operating System (OS)

macOS

### OS Version

macOS Sonoma 14.4.1

### Additional system information

M1 MacBook Pro

### Printer

BambuLab X1C

### How to reproduce

1. Enable "Independent support layer height" under Support > Advanced
2. Choose a material other than your base printing material under "Support/raft interface" under Support > Filament for Supports

### Actual results

When enabling the option above and using a different material for the support interface (for instance, PLA for a PETG print), the support interface material (PLA) is used across the entire layer containing the support interface. Rather than printing the support interface, then changing filament, then printing the rest of the same layer, no filament change is performed and the support interface is not printed on its own at the beginning or end of the layer.

Here's an example of the issue. The first image shows the line types, notice the distinct support interface lines (dark green) and the rest of the "regular" lines for the object like perimeters and infill:
[image]

And here is the filament being used in the same layer, notice the red PLA is being used for the entire layer rather than just for the support interfaces:
[image]

### Expected results

Here is what it should look like, with the red PLA being used only for the support interfaces: [image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-05-03:
As there is no project file attached, what is your purge into infill settings set to?

enz1ey, 2024-05-07:
Sorry, these are proprietary parts so I'm not too keen on posting the project file. I'll slice something generic later today as I'm sure the issue will persist with other models that require supports.

My purge into infill settings aren't used because I don't use a prime tower, I've found that simply adjusting the flushing volumes works just fine for my purposes since I'm typically only using support interface for a single layer, making a prime tower more of a waste.

As part of my troubleshooting, I did enable the prime tower so I could un-check both "Flush into objects' infill" and "Flush into objects' support" and then disabled the prime tower again. This had no effect on the issue. I also tried using "999" in all the flushing volumes to see if perhaps it wasn't flushing enough, but the preview still showed the support interface material being used across the entire layer, and I don't think the flushing volume values really have an effect on the g-code after the purge is complete, so adjusting those values probably doesn't change the materials shown in the preview window.

closed 2024-08-13 by github-actions[bot]