OrcaSlicer/OrcaSlicer #5254: Print a message that tells the filament profile when using Manual Filament Change

rotarucosminleonard, 2024-05-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I attempted my first print using 3 different filament colours. There was no obvious way to know which filament to put in. I messed it out eventually. For my next attempt I analyzed the Gcode file and I noticed the comments. I edited it and manually added a message that posts the filament colour before the pause macro. I used this one because it was coloured: 
`RESPOND TYPE=error MSG='Use White Filament'`

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Print the profile name before the filament change macro.

### Describe alternatives you've considered

Play the gcode and remember/write down the order of the filaments as they are used?

vgdh, 2024-05-04:
You can send the name of the filament to the change macro.

vgdh, 2024-05-04:
FILAMENT_CHANGE_MACRO FILAMENT_NAME='[filament_preset]'

rotarucosminleonard, 2024-05-05:
how is `filament_preset `different from `filament_settings_id` ?

vgdh, 2024-05-05:
> how is `filament_preset `different from `filament_settings_id` ?

I don't know.

closed 2024-08-11 by github-actions[bot]