OrcaSlicer/OrcaSlicer #5258: Segmentation fault on Linux upon launch

Nuc1eoN, 2024-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Archlinux

### Printer

irrelevant

### How to reproduce

Start orcaslicer

### Actual results

Segmentation fault

### Expected results

Should start

### Project file & Debug log uploads

```
orca-slicer                                                                                                                                                                             ~
[2024-05-04 00:00:17.990370] [0x00007632dae8f480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.102: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.102: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.102: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.102: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.103: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.103: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.103: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.103: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.104: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.104: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.105: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.105: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.105: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.105: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.106: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.106: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.107: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.107: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.108: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.108: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): GLib-GObject-CRITICAL **: 00:00:18.108: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.108: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.204: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.206: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.211: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.212: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.212: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.268: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.268: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.275: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.275: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.277: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.277: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.282: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.282: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.287: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.287: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.292: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.292: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.297: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.297: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.301: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:7256): Gtk-CRITICAL **: 00:00:18.301: gtk_widget_set_size_request: assertion 'width >= -1' failed
zsh: segmentation fault (core dumped)  orca-slicer

```

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Actually it used to work. Not sure what happened.

[coredump.zip](https://github.com/SoftFever/OrcaSlicer/files/15205370/coredump.zip)

kvetal, 2024-05-04:
The same situation.
Delete connection to the printer and will get normal start.
Add connect to the printer (klipper+mainsail) and will get crash. After that crash on start.
https://github.com/SoftFever/OrcaSlicer/commit/c2e19b65a8e0f4aaa2d146ed9fea1ef1e47438d6 - Not helped.
Sorry for my english. )

Nuc1eoN, 2024-05-04:
@kvetal I don't think we have the same issue?

The segmentation fault even happens if I remove the folder `.config/OrcaSlicer`.

Or how  do you make it work?

kvetal, 2024-05-05:
> @kvetalЯ не думаю, что у нас одна и та же проблема?
> 
> Ошибка сегментации возникает даже в том случае, если я удалю папку `.config/OrcaSlicer`.
> 
> Или как заставить это работать?

Oh, maybe I was wrong! When I delete a folder, everything starts fine until I add the connection again. Perhaps this is really a different problem. Repeats even when running from docker.

kvetal, 2024-05-05:
> @kvetal I don't think we have the same issue?
> 
> The segmentation fault even happens if I remove the folder `.config/OrcaSlicer`.
> 
> Or how do you make it work?

Oh, maybe I was wrong! When I delete a folder, everything starts fine until I add the connection again. Perhaps this is really a different problem. Repeats even when running from docker.

Nuc1eoN, 2024-05-05:
@kvetal maybe your pr9blem is this https://github.com/SoftFever/OrcaSlicer/issues/4308

Tgey show some workarounds.

But my problem is separate.

lonyelon, 2024-05-06:
This is the same issue as https://github.com/SoftFever/OrcaSlicer/issues/3516.

TLDR: Orcaslicer needs to be linked to webkit2gtk, and the package for it is obviously a dependency.

alive-corpse, 2024-05-06:
On MX Linux 23 AHS I've solved this problem by installing gstreamer1.0-x package. Maybe it can be helpfull for somebody else.

Nuc1eoN, 2024-05-06:
Thank you. Sadly on ArchLinux there are no more gstreamer1.0 packages available.

alive-corpse, 2024-05-06:
> how did you figure it out if I may ask?

It's not my merit. I've found this solution in other issue: https://github.com/SoftFever/OrcaSlicer/issues/3516

jasonslay, 2024-05-09:
installing `gst-plugins-base` on ArchLinux fixed this issue for me as well.

serhii-nakon, 2024-07-15:
I has the same issue on Debian 12 https://github.com/SoftFever/OrcaSlicer/issues/5369#issuecomment-2228668216

lukeallister, 2024-10-02:
I experience the same segfault crash on nixos and on popos (2 different machines)

closed 2025-01-09 by github-actions[bot]

CountingBeeps, 2025-01-09:
Hey github bot, being unusable on my distro is definitely an issue.

serhii-nakon, 2025-01-09:
+

serhii-nakon, 2025-01-09:
Looks like need open new ticket

Nuc1eoN, 2025-01-09:
Intalling `gst-plugins-base` fixed it on ArchLinux, so try that. So in my case it was a downstream issue.