OrcaSlicer/OrcaSlicer #5266: Temperature change in impressions by objects

pganns, 2024-05-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes it is necessary to change the printing temperature (bed, nozzle) in a sequence of prints of several objects or on the same object, with the variation of Z.
You can think about the possibility of carrying out a temperature test at a certain point in the print, as occurs with the temperature calibration tower in the print.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Allow printing temperatures to change when "object printing" is selected. Simplify3D V5 allows a greater degree of freedom and it would be interesting if Orca had this capability.

### Describe alternatives you've considered

Within the "Others" tab, provide the temperature fields in which the printout will be made. The default will be the one set in the filament configuration.

[image]

RXT2020, 2024-05-30:
This is the feature that I currently miss the most.
It would be great if it gets implemented!

closed 2024-09-05 by github-actions[bot]

Big-Channie, 2025-04-29:
Same here. Trying to tune retraction at different temps, and I'm unable to change temp between retraction towers.