OrcaSlicer/OrcaSlicer #5268: When I download the new version and delete the old version, the old version is still there but cannot be opened, and the new version cannot be used as the default opening method for STL

m-i-k-0, 2024-05-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2 22635.3500

### Printer

Bambu Lab P1S

### How to reproduce

It came after the update.

### Actual results

When I download the new version and delete the old version, the old version is still there but cannot be opened, and the new version cannot be used as the default opening method for STL

### Expected results

STL can't be turned on by default, I need to manually open the software to add STL to it。

### Project file & Debug log uploads

I don't know how to do this.
[data.zip](https://github.com/SoftFever/OrcaSlicer/files/15211631/data.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DVSVIDEO, 2024-05-05:
I always use the portable version and this has never happened.

MrOnionAU, 2024-06-26:
I also have this issue and have been trying to find a solution for 4 weeks now...Does anyone have a fix?

closed 2024-10-03 by github-actions[bot]