OrcaSlicer/OrcaSlicer #5272: Tree support ignores independent support layer height setting

mfish38, 2024-05-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 22H2 64bit

### Additional system information

CPU: i7 3930k Display Card: AMD Radeon RX 6900 XT

### Printer

Any

### How to reproduce

1. Enable tree(auto) supports.
2. Have "Independent support layer height" checked.
3. Set a support "top z distance" that is not a multiple of the layer height.
4. Slice
6. Switch to orthographic view and use view widget to switch to side on view.
7. Print

### Actual results

Top Z distance in the sliced output will always be a multiple of the layer height. When printing the support will have zero top z distance and be impossible to remove, or be at least 1 layer and cause droop.

Here is 0.25 layer height with top z distance of 0.124.
[image]

Here is 0.25 layer height with top z distance of 0.125.
[image]

### Expected results

Switching to normal(auto) supports shows correct behavior of having a support top z distance being independent of layer height. 
[image]

### Project file & Debug log uploads

[project_and_log.zip](https://github.com/SoftFever/OrcaSlicer/files/15213780/project_and_log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

poppinfresh69, 2024-05-22:
I'm seeing the same behavior with normal supports. 0.2mm layer, set top Z to 0.28 and rounds the gap up to 0.4. I can see in the preview that it stops the support and then prints two full .2 layers before printing the bridge.

closed 2024-08-28 by github-actions[bot]

Wulfsta, 2024-12-12:
This appears to be an ongoing issue, unfortunate the bot closed it.

excieve, 2025-08-04:
This is still happening in 2.3.0. Moreover, tree supports also behave incorrectly with top and bottom z distances. Possibly due to ignored independent layer height. This makes tree supports barely usable and very erratic.