OrcaSlicer/OrcaSlicer #5273: Bambu Network drivers freezes and crashed on big files

olehmykhalchenko, 2024-05-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1, 2.0.0, 2.1.0

### Operating System (OS)

Linux

### OS Version

Debian 12, Manjaro 23

### Additional system information

Desktop Environment is KDE.
CPU: Ryzen 7 4800h
GPU: Radeon RX5500M
RAM: 32 GB

### Printer

Bambulab X1C

### How to reproduce

1. Install and open Orca slicer
2. Install Bambu Network Driver
3. Open a project with big model/big amount of models
4. Slice plate
5. Print plate

### Actual results

For projects where GCODE is higher then approximately higher then 20mb "Send print job to" works really slow and sometimes freezes. For projects where GCODE is higher then 30mb "Send print job to" freezes instantly after pressing "Print plate" and make OrcaSlicer to crash. 

### Expected results

Succesfuly send a plate to a printer and start printing

### Project file & Debug log uploads

No logs related to this. Project doesnt matter, it happens to every project with GCODE more then 20-30mb

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Works good on Orca Slicer v1.8.1. Tried to move network driver from 1.8.1 to 2.0.0, but It was unsuccesful, because it updates it to newest version right after start.

olehmykhalchenko, 2024-06-14:
Also important to mention this problem is not reproduced with Klipper integration. So its 100% bambu network driver.

ItsJupiter21, 2024-06-14:
Using Bambu Studio does not have the same issue though, it seems like this is caused by the implementation of the bambu drivers.

olehmykhalchenko, 2024-06-24:
Also not working in new 2.1.0 release

travistabbal, 2024-08-22:
I tried the 2.2.0 nightly image. It seems to work better, I have yet to reproduce the issue there and it happens most of the time on 2.1.1 with this file. 

Just in case my issue was different, it would stall out when hitting the print button and loading the dialog box when my Bambu was selected with a large sliced file. This particular gcode file slices to 27.4MB if I save it to disk. 

I'll attach a copy of the project file, in a zip because github doesn't like 3mf files I guess. Source is: 

https://makerworld.com/en/models/539768#profileId-457126   

[4_OrganizerHolder_V00 (Meshed).zip](https://github.com/user-attachments/files/16713701/4_OrganizerHolder_V00.Meshed.zip)

ItsJupiter21, 2024-08-23:
I can also no longer replicate it in the 2.2.0 nightly builds,
tested with a 300% benchy

olehmykhalchenko, 2024-08-29:
It still reproduces for me in the newest nightly build. Version OrcaSlicer_Linux_V2.2.0-dev.AppImage

JacobTheEldest, 2024-09-23:
I am having the same issue with OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta.AppImage on Gnome in EndeavourOS. My printer is a P1S.

jjarven, 2024-10-31:
I might have the same/similar issue, 2.2.0 stable orca but with neptune 4 plus printer.
Sliced the item and printed directly to printer via network - uploading stays at 0%.

I wiresharked all the connections from/to printer - no packets were sent to printer.
Cancelled transfer and tried again, no packets were sent.

Restarted orca, sending to printer works immediately.

[image]
In the image, the "1x4x4 box_0.32mm" were the ones I tried.

Seems longer the orca is left running, the possibility for stuck transfer is increased.

bhaal2, 2024-11-17:
In the later Ubuntu Linux dev release of 2.2.0 it seems to be working now, Also working now is the Home window which never used to work for me in 24.04...

jzobac, 2024-11-26:
Still not working, send dialog freezes. No problem on Mac.

Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble

OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage 

exported 3mf file has 38.8MB; exported gcode has 112MB

wociscz, 2024-12-03:
Workaround - instead of "print plate" option change it to the "send" option. You can send the .gcode directly to the sd card then and start the print from the printer's menu. Device -> MicroSD card -> Model -> print can't be used, because orcaslicer would like to fetch the gcode back to the computer (fetching model information) and send it back to the printer with the "broken" way, but this time it seems to be hanging even for the small files...

romanlex, 2024-12-11:
how long will the problem with downloading files last, it seems to have been going on for years :(

Beep6581, 2025-02-23:
Still valid in v2.3.0-beta.
Duplicate of #3893

Felix14-v2, 2026-10-07:
Could you check the [latest nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/nightly-builds) version and/or #15812?

https://github.com/ClusterM/open-bamboo-networking